FABtotum/fabui-colibri #12: Can't configure Wifi or ETH

ThomasDalgas, 2017-02-24:
Getting PHP error when saving new network settings, and after it saves the Network for both ETH and Wifi i can't configure anything network related, not even after a reboot..
See attached pic.
[image]

daniel-fabteam, 2017-02-24:
Hi Thomas,

Could you elaborate which network settings you were saving. ETH or WiFi?
Which version of FABTotum do you have? You can check this if you go to Maintenance -> System Info
Look for "Hardware"
Also are you using a wifi usb dongle or not?

ThomasDalgas, 2017-02-24:
I'm using the lastest Colibri that was annouced yesterday.
Using the PI3 board, and the internal wifi on that.
It's acually both. After i first choose DHCP on eth, saved it, then lost the connection (makes sense)
I took the power, and it never came up again, just with the yellow LEDs on.
Re-flashed the SD again, and then tried with wifi config, it found my wifi and then pressed save..
After that i got this screenshot i attached. No net connection or anything. Only with the
169.254.2.1..

krios-fabteam, 2017-02-24:
Hi Thomas, 
next time you can try to enter in recovery  mode by turning on the printer without any head installed to preserve system settings.

> After that i got this screenshot i attached. No net connection or anything. Only with the
169.254.2.1..

Could you restart the printer and check if the error is still present?

ThomasDalgas, 2017-02-24:
Tried with Printhead and it's just blinking yellow..
Now i have removed the head, restarted again. It's beeps, blinking red shortly, and now again blinking yellow.
Installed the head again, restarted, and nothing happens except the yellow blinking

BramElderman, 2017-02-24:
I love the whole colibri concept... but so far no success:
in short ...**Network unable to set and Kernal Panic after every reboot**
So I have similar problems, no good network setup possible (where can i set the DNS? it tells me no connection to the internet)  So i tried dual network 169.254 for the ethernet connected unit and wifi module to connect to internet using DHCP that should have DNS... it wil never really connect to wifi.

Jogging and a lot of non-internet connected stuff works really well but only once... until you reboot

Since i have a HDMI monitor connected here are some screenshots 
[image]
[image]
[image]
[image]
[image]

 after setup any reboot will result in kernal panic so no  could select anafter a reboot,.. wipe the card restore content and it works again until you reboot.... 

On the mac the whole proces does not work on safari,.. but works on Firefox...

i saw more people having setup problems might be because of my "older" hardware?

let me know,.. willing to help and get it going

cheers
Bram

daniel-fabteam, 2017-02-24:
@BramElderman great report. You made it to the linux console :)
The files are not found as it seems like a file corruption. Could you try a different sdcard?
Also download the new sdcard_latest. We updated it less then 1h ago.

I suppose you have a RPi1 on your unit and a wifi dongle.

BramElderman, 2017-02-24:
I tried this on 3 different cards,.. always with the same results....
I can also try it on a second FabTotum tonight... but that one has the same hardware...
not sure how i can see the difference... will continue in 2 hours with this thanks for quick response,.. i would love to get this working well

[image]
[image]

daniel-fabteam, 2017-02-24:
Have all the sdcard been 2GB? 
The smallest cards we tested with were 4GB as the first 4 partitions are in total 4GB.

daniel-fabteam, 2017-02-24:
@ThomasDalgas could you try the latest sdcard image? Just download it again.
When doing forced recovery.
1. turn off
2. remove the head
3. turn on and wait for the red lights
4. return the head back
5. continue with recovery

If the network is not going back make sure to un-check "Preserve system settings" so that the network settings are cleared.

BramElderman, 2017-02-24:
I believe I had a mix of 4 GB and 2 GB cards but I'll retry with the new download on the 4 GB card and let you know in an hour and a half, what was changed in the new build?

Groet
Bram

Verstuurd vanaf mijn iPhone

BramElderman, 2017-02-24:
Used another 4G card downloaded the new version but unfortunately no difference... ;-(

The forced recovery you offered to Thomas worked flawless,... (that was new to me) however it brings the device in the same state that does not allow to connect to wifi or have a working fixed IP number on ethernet (including DNS setting) .... it simply does not connect to the wifi see video enclosed. (as zipped movie)
[IMG_7887.MOV.mov.zip](https://github.com/FABtotum/fabui-colibri/files/800499/IMG_7887.MOV.mov.zip)
Also at reboot we get kernal panic.... and we need to do recovery

the rest feels good and very solid but this is impossible to get started...

daniel-fabteam, 2017-02-24:
Let's split this into another issue.
The WiFi dongle connection problem is known and is mentioned as "Know Issue". Will be fixed soon.

The boot issue where the partitions are not found is a different thing. I will open it as new issue and we will continue there. So we don't mix multiple bugs in one issue thread.

BramElderman, 2017-02-24:
OK cool let me know if the Ethernet setup including dns will be addressed by the wireless adapter/networking issue... or is that a separate issue... 

Keep up the good work! although struggling now this feels a huge improvement over the old software structure !!

daniel-fabteam, 2017-02-27:
We have released a new [sdcard image](https://trello.com/c/XDl81yUZ/131-beta-github-issues) so could you try it out and report the issue is fixed?

dr4Ke, 2017-02-27:
Good news @daniel-fabteam. How do we access this image, though? Do we need an account on trello.com?

daniel-fabteam, 2017-02-27:
Sorry for the wrong link. I've updated it. 

It's the same one as on first page of this repository the you used to download the first sdcard image.

BramElderman, 2017-02-27:
Hey Daniel thank you

I will download it immediately when I'm back home in about an hour and a half, should it be downloaded from github or can we downloaded from the fabtotum site as the latest build??

Whatever the result is I'll let you know immediately when I have installed it and tested it the first time thanks!
I'm sure I will be back in a couple of hours with the results!

Groet
Bram

Verstuurd vanaf mijn iPhone

daniel-fabteam, 2017-02-27:
The [sdcard_lates.zip](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip) link is the same everywhere and is always kept up-to-date with the latest sdcard release. I've just put the link here to make it easier to click and download.
You can bookmark it and when you hear there is a new one just download it again.

PaoloP74, 2017-02-27:
In the latest release of the gateway value is null.

BramElderman, 2017-02-27:
Unfortunately nothing really changed,…

Directories are still not created..

Wifi Connection is still not connecting or stored…

Reboot results in kernel panic every time…

Bummer,…
Here are the screenshots…

krios-fabteam, 2017-02-27:
Hi @BramElderman,
take a look here #20

daniel-fabteam, 2017-02-28:
@PaoloP74 gateway is left empty for not configured network interfaces. Is not a bug.

By the way we have release a new sdcard which should fix several issues.
Make sure to read the guide [how to format](https://github.com/FABtotum/fabui-colibri/wiki/SDCard-formatting) the sdcard.

daniel-fabteam, 2017-02-28:
@ThomasDalgas Could you try the new sdcard image and follow the our new [formatting guide](https://github.com/FABtotum/fabui-colibri/wiki/SDCard-formatting) while making it?

BramElderman, 2017-02-28:
I still cannot connect the wireless,... i do have a # in the password that might cripple the json string in the parser...

daniel-fabteam, 2017-02-28:
@BramElderman Good point. We will check special characters tomorrow and see if they cause problems.
Maybe from now on you can use update in the menu instead of making a new sdcard. As it seems the sdcard is working now.

BramElderman, 2017-02-28:
Yes the card is working and the Colibri really feels the promised lightweight bird...
Great Job,... i will check the software-update option from now in...
The only thing missing now is the LASER option in "Make" chapter ,.. i can only print mill or scan.... aultough i have the laser head installed...

BramElderman, 2017-02-28:
Ethernet is working fine now,.. however i think there should have been a manual DNS server line to set these... it is working so i assume you have baked-in 8.8.8.8 or something but that doesn't work everywhere,.. you might give a default if left empty but give the user the option to have their in-house DNS

BramElderman, 2017-02-28:
Did some additional testing i am pretty confident that the # in my password (although aperfect legal character in ssid password) is confusing the parser.
**To test i setup a PersonalHotspot** on my phone with a plain-tekst pasword,.. it connected perfectly and when trying to the "#" containing network it failed and this time when switching bach even with a PHP error that seems to have a error handler problem
[image]

daniel-fabteam, 2017-03-01:
@BramElderman Hello. In the latest updates there should be wifi fixes. I have tried passwords with # and { but it worked in the end. Also settings are encoded before saving so special characters should be safe. There was a different issue though which is fixed now.

Are there more special characters in your password?
"<>'& are the only forbidden characters by the wifi standard and now cannot be entered in the password input fields.

Also Ethernet hotplug support is added so you should be able to connect directly to your home network over wire and access your fab unit with "fabtotum.local" address.

BramElderman, 2017-03-01:
OK was home late so sorry for the long wait...
Software update on the device was not possible because "no internet connection" although i was connecting through ethernet with a manual IP (still the lack of DNS entry puzzles me why the fabtotum has no internet connection, please make the DNS line available in network settings, or clarify how the unit should resolve connections)
So i did a "new card installation" on the working 8G SDcard
Unfortunately Wireless failed again,  with has no other charakters then #... so you suggested fix could have worked,.. but it might also be the "old wifi USB module??"
so for now i would stick with ethernet and after some puzzling and reboots the device was capable of resolving internet and the sofrware update mentioned it was up to date.!!
[image]

daniel-fabteam, 2017-03-02:
@BramElderman I will my unit with the old dongle (and RPi1) today and see if I have any issues. Did a connect/disconnect cycle with it now and seems to work.
Could you write a template of your password here (replace every character with a different one of the same type, keep the # where it is) or send that particular piece of info to my email dk@fabtotum.com 
If all goes well DNS settings will be added in the next update.

BramElderman, 2017-03-02:
YES you are a hero,.. !! DNS settings are in !!
my password is Arya599#nuwg I will test with my wifi dongle in 10 minutes... although I tend to use the EHERNET only but it should work for other users...

BramElderman, 2017-03-02:
Hmm the wifi now brings up a PHP problem,... I will shutdown and start again
[image]

daniel-fabteam, 2017-03-02:
I tried your password on my wifi router and it worked. At this point it shouldn't be password issue.
However the PHP error suggests there might be a different problem.

What encryption scheme are you using for your wifi?

There is a plugin **System logs**. Install and activate it. You will have a new menu item **SYSTEM LOGS*. Go there, press download and attache the resulting `archive.tar.gz` here.

BramElderman, 2017-03-02:
It is getting bizarre,... the scanning of wifi no longer works...  have no clou why??
The ethernet with fixed IP is in service,.. can I assume they cn work at the same tim or is its just one of them???
Here are the logs...

[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/815235/archive.tar.gz)

BramElderman, 2017-03-02:
oh by the way I discovered that I have 2 different wifi usb plugs,.. one without a button and one with a tiny button on the top,... they came with the 2 Fabtotums I ave got about 3 months after each other..
just so you know

daniel-fabteam, 2017-03-03:
Not sure how but some of the logs are in Japanese!

The dongle without the small button is the older model. But both models should work by now.
We'll need to update the plugin as two pieces of logs are missing.

Could you try to connect to wifi again and when it fails login to your unit over ssh.
`ssh root@fabtotum22.local` no password needed (for now) and send me the output of

    ifconfig

    iwconfig

daniel-fabteam, 2017-03-03:
Do you have an apple keyboard connected to the unit? There 3 more usb devices in the logs then expected. Maybe that could affect some things. Using is safer at this point and you'll be able to copy paste the output. No need to take pictures.

BramElderman, 2017-03-03:
OK i wil continue tonight with this all:

YES i have keyboard connected to do CLI stuf with the old versions in case the network was not there or didnt work,.. the ethernet seems stable now so i can disconnect en do it all over ssh

My encryption scheme is WPA2-personal btw…

BramElderman, 2017-03-03:
Tonight did the available update and tested both WiFi dongles the one with the and without the button... in both cases the scan is not delivering a list of ssid networks,.. the one without the button simply shows nothing, the one with the  button on the unit gives PHP error 
see logfiles enclosed
[Wifi scan fail and PHP error only with wifi module with button.gz](https://github.com/FABtotum/fabui-colibri/files/817854/Wifi.scan.fail.and.PHP.error.only.with.wifi.module.with.button.gz)
[Wifi scan fail with wifi module without button.gz](https://github.com/FABtotum/fabui-colibri/files/817855/Wifi.scan.fail.with.wifi.module.without.button.gz)

BramElderman, 2017-03-03:
The one with the button (PHP error):

ifconfig
eth0      Link encap:Ethernet  HWaddr B8:27:EB:0A:EC:1B  
          inet addr:172.16.100.22  Bcast:172.16.255.255  Mask:255.255.0.0
          inet6 addr: fe80::ba27:ebff:fe0a:ec1b/64 Scope:Link
          inet6 addr: fd9c:3f7e:81b8:1:ba27:ebff:fe0a:ec1b/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4662 errors:0 dropped:13 overruns:0 frame:0
          TX packets:970 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:787881 (769.4 KiB)  TX bytes:653261 (637.9 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:45 errors:0 dropped:0 overruns:0 frame:0
          TX packets:45 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:10518 (10.2 KiB)  TX bytes:10518 (10.2 KiB)

[root@fabtotum22 ~] iwconfig
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.

BramElderman, 2017-03-03:
The PHP error:
A PHP Error was encountered
Severity: Warning

Message: Invalid argument supplied for foreach()

Filename: controllers/Settings.php

Line Number: 161

Backtrace:

File: /usr/share/fabui/application/controllers/Settings.php
Line: 161
Function: _error_handler

File: /usr/share/fabui/index.php
Line: 316
Function: require_once

A PHP Error was encountered
Severity: Warning

Message: Invalid argument supplied for foreach()

Filename: controllers/Settings.php

Line Number: 174

Backtrace:

File: /usr/share/fabui/application/controllers/Settings.php
Line: 174
Function: _error_handler

File: /usr/share/fabui/index.php
Line: 316
Function: require_once

BramElderman, 2017-03-03:
Th wifi module without button does not throw an error but scan results in nothing....

Welcome!
[root@fabtotum22 ~] 
[root@fabtotum22 ~] 
[root@fabtotum22 ~] 
[root@fabtotum22 ~] ifconfig
eth0      Link encap:Ethernet  HWaddr B8:27:EB:0A:EC:1B  
          inet addr:172.16.100.22  Bcast:172.16.255.255  Mask:255.255.0.0
          inet6 addr: fe80::ba27:ebff:fe0a:ec1b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1032 errors:0 dropped:3 overruns:0 frame:0
          TX packets:498 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:320158 (312.6 KiB)  TX bytes:234761 (229.2 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:37 errors:0 dropped:0 overruns:0 frame:0
          TX packets:37 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:10985 (10.7 KiB)  TX bytes:10985 (10.7 KiB)

[root@fabtotum22 ~] 
[root@fabtotum22 ~] 
[root@fabtotum22 ~] iwconfig
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.

daniel-fabteam, 2017-03-03:
@BramElderman I'm terribly sorry for all the trouble with the WiFi you are having. And am grateful you are still with us trying to solve this issue.

The root of the PHP error is that this command does not return a proper value
`sh /usr/share/fabui/ext/bash/get_net_interfaces.sh`

Scanning results are gather with this script
`python /usr/share/fabui/ext/py/scan_wifi.py`

Configuring the WiFi is done like this
`sh /usr/share/fabui/ext/bash/set_wifi -i wlan0 -s "YourSSID" -p "YourPassword" -D`

If you could execute all of the above on the unit using ssh and attach the output. Maybe we'll finally find what is the problem.
Not all error messages are passed all the way to the UI so maybe this will help show what is really going wrong.

BramElderman, 2017-03-04:
Don't worry actually i don't need wifi i normally have the Fabtotums Ethernet connected,.. but i want the Colibri OS to be perfect so it should work even if i would not use it.

Configuring using CLI did not work,... rest did see below

Offending key for IP in /Users/bram/.ssh/known_hosts:4
Are you sure you want to continue connecting (yes/no)? yes
Welcome!
[root@fabtotum22 ~] 
[root@fabtotum22 ~] 
[root@fabtotum22 ~] 
[root@fabtotum22 ~] sh /usr/share/fabui/ext/bash/get_net_interfaces.sh
{
  "eth0" : {
    "driver" : "smsc95xx", 
    "mac_address" : "b8:27:eb:0a:ec:1b",
    "ipv4_address" : "172.16.100.22/16",
    "ipv6_address" : "fd9c:3f7e:81b8:1:ba27:ebff:fe0a:ec1b/64",
    "address_mode" : "static",
    "gateway" : "172.16.0.1"
  },
  "wlan0" : {
    "driver" : "rtl8192cu", 
    "mac_address" : "00:0f:13:96:04:e6",
    "ipv4_address" : "",
    "ipv6_address" : "",
    "address_mode" : "dhcp",
    "gateway" : "",
    "wireless" : {
      "can_be_ap" : "yes",
      "mode" : "auto"
    }
  }
}
[root@fabtotum22 ~] 
[root@fabtotum22 ~] python /usr/share/fabui/ext/py/scan_wifi.py
{}
[root@fabtotum22 ~] 
[root@fabtotum22 ~] 
[root@fabtotum22 ~] python /usr/share/fabui/ext/py/scan_wifi.py
{}
[root@fabtotum22 ~] 
[root@fabtotum22 ~] sh /usr/share/fabui/ext/bash/set_wifi -i wlan0 -s "Landgoed Soelen WiFi" -p "Aega800#cied" -D
sh: can't open '/usr/share/fabui/ext/bash/set_wifi'
[root@fabtotum22 ~] cd /usr/share/fabui/ext/bash/
[root@fabtotum22 bash] ls -la
total 26
drwxrwxr-x    2 root     root           330 Mar  3 17:28 .
drwxrwxr-x    3 root     root             4 Feb 23 11:04 ..
-rw-rw-r--    1 root     root           212 Mar  3 17:28 disconnect_wifi.sh
-rw-rw-r--    1 root     root           285 Mar  3 17:28 emergency.sh
-rw-rw-r--    1 root     root           315 Mar  3 17:28 get_dns.sh
-rw-rw-r--    1 root     root          3802 Mar  3 17:28 get_net_interfaces.sh
-rw-rw-r--    1 root     root           470 Mar  3 17:28 internet.sh
-rw-rw-r--    1 root     root           642 Mar  3 17:28 recovery.sh
-rw-rw-r--    1 root     root           542 Mar  3 17:28 resetController.sh
-rw-rw-r--    1 root     root          1458 Mar  3 17:28 scan_wifi.sh
-rw-rw-r--    1 root     root          2760 Mar  3 17:28 set_ethernet.sh
-rw-rw-r--    1 root     root          1245 Mar  3 17:28 set_hostname.sh
-rw-rw-r--    1 root     root          1325 Mar  3 17:28 set_time_zone.sh
-rw-rw-r--    1 root     root          5929 Mar  3 17:28 set_wifi.sh
-rw-rw-r--    1 root     root          3269 Mar  3 17:28 totumduino_manager.sh
-rw-rw-r--    1 root     root           321 Mar  3 17:28 utilities.sh
[root@fabtotum22 bash]

daniel-fabteam, 2017-03-04:
My bad. It should be `sh /usr/share/fabui/ext/bash/set_wifi.sh`

BramElderman, 2017-03-04:
Okido,... better but not good... however i do realise that there are 2 spaces (%20) in the SSID name,... "Landgoed Soelen WiFi" i tried both in the CLI but without succes
Space is legal,... but could be wrongly parsed...

Logfiles and the rest below
[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/818946/archive.tar.gz)

After this the scan seemed to work but still no connection... and PHP error in the web dialog

Are you sure you want to continue connecting (yes/no)? yes
Welcome!
[root@fabtotum22 ~] sh /usr/share/fabui/ext/bash/set_wifi.sh -i wlan0 -s "Landgoed Soelen WiFi" -p "Aega800#cied" -D
sh: Soelen: unknown operand
no dnsmasq range
no hostapd conf
dhcpcd[1565]: dhcpcd not running
no dnsmasq range
no hostapd conf
no dnsmasq range
no hostapd conf
Successfully initialized wpa_supplicant
Line 6: unknown global field 'Passphrase must be 8..63 characters'.
Line 6: Invalid configuration line 'Passphrase must be 8..63 characters'.
Failed to read or parse configuration '/etc/wpa_supplicant/wlan0.conf'.
run-parts: /etc/network/if-pre-up.d/wpa_supplicant: exit status 1
Failed to bring up wlan0.
[root@fabtotum22 ~] sh /usr/share/fabui/ext/bash/set_wifi.sh -i wlan0 -s "Landgoed%20Soelen%20WiFi" -p "Aega800#cied" -D
no dnsmasq range
no hostapd conf
dhcpcd[1612]: dhcpcd not running
no dnsmasq range
no hostapd conf
no dnsmasq range
no hostapd conf
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
ioctl[SIOCSIWAP]: Operation not permitted
dhcpcd[1699]: version 6.6.7 starting
dhcpcd[1699]: wlan0: adding address fe80::34ea:69ce:7cf3:1e85
dhcpcd[1699]: wlan0: waiting for carrier
dhcpcd[1699]: timed out
dhcpcd[1699]: exited
Failed to bring up wlan0.

daniel-fabteam, 2017-03-04:
I think we are finally at the root of the problem!
The SSID and Password were not "space" safe.

I've attached a fixed version of `set_wifi.sh`. It will be available in the new updates next week.
[set_wifi.sh.zip](https://github.com/FABtotum/fabui-colibri/files/818961/set_wifi.sh.zip)

You can upload it to your unit with (unzip it first on your computer):
`scp set_wifi.sh root@fabtotum22.local:/usr/share/fabui/ext/bash/`

Now run the command to `set_wifi.sh` as before and it should work. Have made it **paranoidly** space safe.

Afterwards you must revert back to the original by doing this:
`rm /mnt/live/mnt/changes/usr/share/fabui/ext/bash/set_wifi.sh`
If you don't, the next update will not be able to replace it.

BramElderman, 2017-03-06:
Sorry it took a little longer was a bit away this weekend,... but back on track now !

I think you nailed it,... ;-)

it worked see log below and afterwards in the GUI it showed it in the interface as well

[root@fabtotum22 ~] sh /usr/share/fabui/ext/bash/set_wifi.sh -i wlan0 -s "Landgoed Soelen WiFi" -p "Aega800#cied" -D
no dnsmasq range
no hostapd conf
dhcpcd[1726]: dhcpcd not running
no dnsmasq range
no hostapd conf
no dnsmasq range
no hostapd conf
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
ioctl[SIOCSIWAP]: Operation not permitted
dhcpcd[1775]: version 6.6.7 starting
dhcpcd[1775]: DUID 00:01:00:01:20:4f:de:98:00:0f:13:96:04:e6
dhcpcd[1775]: wlan0: IAID 13:96:04:e6
dhcpcd[1775]: wlan0: soliciting a DHCP lease
dhcpcd[1775]: wlan0: soliciting an IPv6 router
dhcpcd[1775]: wlan0: offered 172.16.0.116 from 172.16.0.1
dhcpcd[1775]: wlan0: leased 172.16.0.116 for 86400 seconds
dhcpcd[1775]: wlan0: adding route to 172.16.0.0/16
dhcpcd[1775]: wlan0: adding default route via 172.16.0.1
dhcpcd[1775]: forked to background, child pid 1803
no dnsmasq range
no hostapd conf
[root@fabtotum22 ~] 

[image]

BramElderman, 2017-03-06:
And as expected after removing i get the PHP error back.... so you can implement it in the next build...

daniel-fabteam, 2017-03-06:
Great news!
I will be included in today's update.

This issue can be finally closed.

closed 2017-03-06 by daniel-fabteam