FABtotum/fabui-colibri #13: A couple of bugs..

ugo3d, 2017-02-24:
After a calibration i tried to print but i got this error:
[image]
then i make a reboot and it stopped here:
[image]
but with no rebooting.
making the sd again...
ciao

AndrewKillian, 2017-02-24:
I had the same no reboot issue tried remaking the sd but no luck so far

daniel-fabteam, 2017-02-24:
Could you create a separate issue for the two bugs you found. Is easier to track and handle.
In the new issues we will continue the discussion.

Thanks for the help.

krios-fabteam, 2017-02-24:
The first issue is fixed in the latest FABUI update.
Please confirm if it really was fixed.

AndrewKillian, 2017-02-24:
I never got the first issue just the restarting and I was on Colibri then

closed 2017-02-24 by kriosmane

krios-fabteam, 2017-02-24:
Fix ready in repository, will be available with the next update

commit 7787b7c5 (2017-02-24): Issue fix: #13