FABtotum/fabui-colibri #15: unable to login

lahines100, 2017-02-24:
After loading the software and receiving the steady blue light, I was unable to connect to the printer. I was using a cable connection. 
The original FABUI still works.

daniel-fabteam, 2017-02-25:
@lahines100 which version of unit do you have? Is it a RPi1 or RPi3 base?

lahines100, 2017-02-26:
I solved the issue today by reformatting the 8GB memory chip to 2GB and 6GB. When the new software was loaded into the 2GB section I could then log onto the FABtotum. It appears the size of the memory card does matter. Also, I upgraded to the new core a few weeks ago.
Thanks Daniel.

daniel-fabteam, 2017-02-26:
@lahines100 Size should only matter in the sense that is has to be big enough which is 4GB.
Maybe the sdcard was formatted into unsupported filesystem.

Could you try to format the sdcard again fully (8GB no subpartitions) but make sure to select FAT32 filesystem.

daniel-fabteam, 2017-02-27:
As the issue seems to be fixed I'll close it.
Make sure to checkout the new sdcard as we released today. The intention is to release a new sdcard image every day as long as there are open issues.

closed 2017-02-27 by daniel-fabteam