FABtotum/fabui-colibri #19: Add support for backup and restore of user-data and settings

rcmaniac25, 2017-02-25:
Currently, and on the prior OS, the method of backing up and restoring data involved backing up or restoring the actual binary, bit-for-bit, of the SD card.

This necessitated turning off the FAB and removing the side panel, as well as taking up potentially massive amounts of disk space (ex. It's cheaper for me to buy a 32GB card then an 8GB one, but backups end up being 32GB because of that).

A backup and restore function would reduce the backup to only the user data and settings and could/should be possible to backup without shutdown/restart of FAB. Restore, I'd love to not need to restart, but it would probably be better for a restart.

Expected data to backup, at minimum:
- Users and their settings
- GCode (the files and database entries)
- System settings
- Print/scan/mill/laser history

IT2BE, 2017-03-02:
But all options should be... Options :)

E.g. a checkbox to select what you need/want to backup.

daniel-fabteam, 2017-03-02:
Could some of you try the new recovery as it allows to restore system AND user settings from previously damaged system. It has what you describe. Checkbox based wizard of what to restore or not.

1. Turn off the unit.
2. Remove the head
3. Turn on the unit...wait until red lights
4. Connect to the unit over Ethernet cable like the first time you boot with 169.254.1.2
5. Uncheck "Preserve system settings"
    This will wipe the system partitions and preserve user settings.
6. When system restoring is finished you will be presented with a restore wizard to select what you want to restore in detail.

daniel-fabteam, 2017-03-02:
Here is a [link](https://github.com/FABtotum/fabui-colibri/wiki/Recovery) to our fabui-colibri wiki which describes the recovery in more detail.

PaoloP74, 2017-03-29:
How migrate settings and user data from the standard FabUI to the new Colibrì-FabUI?

rcmaniac25, 2017-06-10:
Recovery feature is nice for same-FAB, always the same system setups. What if I want to have multiple setups or switch between FABs? I still need to be able to backup and restore to a zip or something similar.