FABtotum/fabui-colibri #20: Cannot boot up after shutting down FABtotum

LeiEng, 2017-02-25:
After upgrading to the new Colibri OS I cannot ever boot up my FABtotum again after shutting it down.

This is what I do:

- Boot up my FABtotum with a newly created SD-card (with files extracted from sdcard_latest.zip)
- Open a web browser and go to 169.254.1.2
- Do all neccesary setup (create my personal account, setup language/timezone, sign in to FabUI)
- At this point I can use the FABtoum, move the head around and so on
- I now select "Shut down" from the top menu

At this point I get the message:
    "Shutdown in progress"
    "Please wait..."

I never get anything more, but the FABtotum stops answering pings to 169.254.1.2. The output from the Raspberry Pi3 (I have it connected to a screen) also goes black. After 5 minutes without any more messages I turn the FABtotum off.

When I startup the FABtotum again it never gets online. It does not answer ping to 169.254.1.2 either.

A picture of the screen output from the Raspberry Pi3 is attached.
 
[image]

BramElderman, 2017-02-25:
+1 similar to me embedded in my comment of "Can't configure Wifi or ETH #12" also ending in kernel panic.. seems better to have that in this separate ticket.

daniel-fabteam, 2017-02-27:
Hi,

We have released a new [sdcard image](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip) so you can try to make a new sdcard and try if it fixed the issue.

LeiEng, 2017-02-27:
Thanks for the update!

Unfortunately it did not help in my case, I could not boot again after shutting down the FABtotum.
I have a Raspberry Pi 3 and a Totumduino v2.

Do you want me to attach the screendump of the Kernel Panic?

Is there anything else I can supply to you which would help you to track down the cause of this problem?

daniel-fabteam, 2017-02-27:
@LeiEng We haven't had a kernel panic in moths. Could you try to use a different sdcard?

The system should not corrupt even if you pull the plug while running a task. So especially if you use the shutdown button it should not do anything funny.

Let's see if you have the same issue with a different sdcard.

LeiEng, 2017-02-27:
I have now tried with three completely different SD-cards, but I still get the same problem with them all.

The first card is a brand new 16GB Sandisk Ultra Micro-SDHC, model number SDSQUNC-016G-GN6MA
The second card is an 8GB Sandisk Ultra Micro-SDHC, probably the same model as the first one.
The third card is a 4GB Kingston Micro-SDHC, I don't know the model number for it.

dr4Ke, 2017-02-27:
Got a kernel panic too. Running on the crowdfunding version of the fabtotum, so not the same hardware as @LeiEng.

With a monitor, I can see on it that it could not mount some partitions. And there are some files that can't be created. Screenshot:

[image]

I'm using a 16GB SanDisk SDHC. On computer, I can see the original partition, shrinked to 512MiB, and a 3MiB one with no type set.

dr4Ke, 2017-02-27:
I succeeded to boot a second time. Here's what I did:

- remove all partitions on my sdcard
- create one partition, on the whole sdcard, with a W95 FAT32 (LBA) partition type
- copied the sdcard_latest content on it
- booted on the fabtotum
- run the initial setup, with a monitor attached
- at each step, I looked at the sdcard partition table
- it add 6 partitions after the re-partitioning and after shutdown

So it seems to boot fine for me now.

krios-fabteam, 2017-02-27:
@dr4Ke great, 
maybe we should improve the installation guide by adding more specific details

@BramElderman could you try as @dr4Ke did

BramElderman, 2017-02-27:
Yes will try but how can i look at the sdcard partition table??
BTW i am running on the IndiGogo version hardware so RPi1

krios-fabteam, 2017-02-27:
Actually if your unit boot and you are able to do the first installation is not needed to look at the sdcard partitions as it means that everything is ok. 
Btw to look if these partitions are there you have to insert the sdcard into a pc with a linux distribution or into a mac. On windows systems only the "boot" partition is accessible

BramElderman, 2017-02-27:
Yes the initial setup runs fine,...  (except for the wifi not being stored, but for me that might be because there is no user directory for storing the setting) I run everything on the Mac so i can have a look... however cannot have a look in between because that requires the machine to shutdown... 
however it ends with a kernel panic,.. are you saying i should just wait 5 minutes after the panic? and it will recover by crating directories???

BramElderman, 2017-02-27:
So my card is a 4G card... this is what my partition looks like if i look at it from the unit
(still no user dir)
[image]

daniel-fabteam, 2017-02-27:
Please run  `blkid` and take a picture of the output. That will show the real partitions. This shows only the mounted ones.

daniel-fabteam, 2017-02-27:
And also this command
`echo "p " | fdisk /dev/mmcblk0`

Please :)

BramElderman, 2017-02-27:
[image]
[image]

sure here you go

LeiEng, 2017-02-27:
I get the same otuput:

[root@fabtotum ~] blkid
/dev/loop/12: TYPE="squashfs"
/dev/loop/11: TYPE="squashfs"
/dev/loop/10: TYPE="squashfs"
/dev/loop/9: TYPE="squashfs"
/dev/loop/8: TYPE="squashfs"
/dev/loop/7: TYPE="squashfs"
/dev/loop/6: TYPE="squashfs"
/dev/loop/5: TYPE="squashfs"
/dev/loop/4: TYPE="squashfs"
/dev/loop/3: TYPE="squashfs"
/dev/loop/2: TYPE="squashfs"
/dev/loop/1: TYPE="squashfs"
/dev/loop/0: TYPE="squashfs"
/dev/loop12: TYPE="squashfs"
/dev/loop11: TYPE="squashfs"
/dev/loop10: TYPE="squashfs"
/dev/loop9: TYPE="squashfs"
/dev/loop8: TYPE="squashfs"
/dev/loop7: TYPE="squashfs"
/dev/loop6: TYPE="squashfs"
/dev/loop5: TYPE="squashfs"
/dev/loop4: TYPE="squashfs"
/dev/loop3: TYPE="squashfs"
/dev/loop2: TYPE="squashfs"
/dev/loop1: TYPE="squashfs"
/dev/loop0: TYPE="squashfs"
/dev/mmcblk0p1: LABEL="BOOT" UUID="FB52-FA69" TYPE="vfat"
[root@fabtotum ~] echo "p" | fdisk /dev/mmcblk0

The number of cylinders for this disk is set to 15680.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Disk /dev/mmcblk0: 7580 MB, 7948206080 bytes, 15523840 sectors
15680 cylinders, 198 heads, 5 sectors/track
Units: cylinders of 990 * 512 = 506880 bytes

Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1    0,130,3     65,197,5          8192    1056640    1048449  511M  b Win95 FAT32
/dev/mmcblk0p2    2,13,4      8,54,2            2048       8191       6144 3072K  0 Empty
Partition 2 does not end on cylinder boundary

Command (m for help): [root@fabtotum ~]

daniel-fabteam, 2017-02-27:
Thank you.

First partition holds the boot files and recovery. That one is there otherwise you'll not be able to boot.
Second partition is kind of there and it holds the working system copy.
And the rest is missing completely.

There seems to be something unexpected when the sdcard is created on a Mac. We will need to investigate this i more detail.

I haven't use a Mac so I don't know much about the tools installed there.

Are `fdisk` and `dd` commands available in a mac terminal?

BramElderman, 2017-02-27:
yes they are

BramElderman, 2017-02-27:
it is basically freeBSD

LeiEng, 2017-02-27:
I am creating the SD-card from a Windows 10 PC. Formatting is done with "SDFormatter V4.0" and unpacking is done with "7-Zip 15.14 [64-bit]"

daniel-fabteam, 2017-02-27:
This might be dangerous if you miss the partition number. So be warned.
${SDCARD_IMG} will need to be replace by the device your sdcard is identified by.
`blkid` is a good way to identify the correct device in linux.

The following commands are what we use to create a new sdcard. First `dd` destroys the existing partition table of the sdcard. Again, if you miss the device you will nuke your hard drive.
The second command creates one FAT32 W95 partition of the full sdcard size.

> dd if=/dev/zero of=${SDCARD_IMG} bs=1M
> 
> echo "
> o
> n
> p
> 1
> 
> 
> t
> b
> w
> 
> " | fdisk ${SDCARD_IMG}
>
> mkfs.vfat -n BOOT ${SDCARD_IMG}

After this the sdcard is clean and ready for the sdcard_latest.zip content.

daniel-fabteam, 2017-02-27:
@LeiEng Can you check when you are formatting the sdcard that the filesystem is FAT32? Newer windows tend to format removable devices to ExFAT which is not supported by RPi or the tools we used on it.

LeiEng, 2017-02-27:
It seems to be FAT32:

C:\Users\LeiEn>chkdsk s:
The type of the file system is FAT32.
Volume BOOT created 2017-02-27 22:49
Volume Serial Number is 243D-6B62
Windows is verifying files and folders...
File and folder verification is complete.

Windows has scanned the file system and found no problems.
No further action is required.
    7 753 728 KB total disk space.
           32 KB in 1 hidden files.
           32 KB in 1 files.
    7 753 632 KB are available.

       32 768 bytes in each allocation unit.
      242 304 total allocation units on disk.
      242 301 allocation units available on disk.

[image]
[image]

daniel-fabteam, 2017-02-27:
@BramElderman We will check how to create correct sdcards on a Mac and get back to you with a step by step guide.
Thanks for your help.

BramElderman, 2017-02-27:
OK cool thanks, i can try an do your instruction as above on a "risk-less" mac... if needed or come back to me tomorrow...

daniel-fabteam, 2017-02-27:
@LeiEng We will try to create an sdcard with the same OS and tools as you did and see what happens.
As soon as we figure out the root of this problem and fix it everyone can start testing the new FABUI.

krios-fabteam, 2017-02-28:
Hi guys, 
we have wrote a step by step guide for the SD card formatting in the [wiki ](https://github.com/FABtotum/fabui-colibri/wiki/SDCard-formatting)section,  please take a look into it.
We hope this can help you fix the issue

BramElderman, 2017-02-28:
Hi Krios 
Thanks for the good description on the how to prepare the SD card for Colibri file installation... it is handy for everyone in the future, however it was about what i did and unfortunately doesn't solve the reboot or wifi issues perhaps because the correct structure is not created...
[image]
[image]

Also created a screenrecoring of the wifi not sticky but that might be related to not having a user data directory...

[Setup Wifi on Colibri Fabtotum480.zip](https://github.com/FABtotum/fabui-colibri/files/807518/Setup.Wifi.on.Colibri.Fabtotum480.zip)

LeiEng, 2017-02-28:
Thanks for the SD card formatting instruction!

I used MiniTool Partition Wizard Free Edition and now I can shutdown/reboot my FABtotum with Colibri OS without any problems!

I think that it would be a good idea to update the **Installation** section at [https://github.com/FABtotum/fabui-colibri](https://github.com/FABtotum/fabui-colibri) with a link to the wiki article about how to format the SD card.

BramElderman, 2017-02-28:
how bizarre...I did it on a 4Gb card i will try is again on a 8Gb card....

daniel-fabteam, 2017-02-28:
@BramElderman Could you try to format it on a different OS if you have one? We confirmed that windows 10 and SD formatter V4 makes the sdcard wrong.
The first partition must not have any space before it, otherwise the result is when you are getting.

daniel-fabteam, 2017-02-28:
@BramElderman What operating system are you using to format the sdcard?

daniel-fabteam, 2017-02-28:
@LeiEng Great new. There is a link to the full wiki but yes now it also has a link on Format to make more visible.

BramElderman, 2017-02-28:
VICTORY! i used a 3rd SD card this time it was an 8Gb card,... did the same process as before but now got **no failure of creation** of the user directory....(previous was a 4Gb SDHC speed4)
The original messaging was, that it was so lightweight that it fitted any card.. so i grabbed a card lying around to experiment.. it needs 8G or bigger apparently... although i am not sure

Still have the Wifi and network problem,... cannot set the SSID,.. and cannot set the network manually... **BUT I CAN REBOOT WITHOUT PANIC!**

[image]

daniel-fabteam, 2017-02-28:
Finally the boot looks as it should. I'm using 4GB cards without issues. 
The ones I have are Transcend SDHC 4GB Class 4. Also an Kingston SDHC 8GB Class 4.

Let's close this issue as the boot and reboot works.

closed 2017-02-28 by daniel-fabteam

daniel-fabteam, 2017-02-28:
@BramElderman Could you check the partition table of your working and non-working sdcards?

If I recall correctly you have `fdisk` in your terminal. Insert and sdcard, (follow the formatting guide if you to see how to identify the device) and run `fdisk /dev/identified_sdcard_device`

Then type **p** and **ENTER**. This should list all partitions and their sizes.
Do this for both sdcards and post the result here.

BramElderman, 2017-02-28:
Will do in one hour 

Groet
Bram

Verstuurd vanaf mijn iPhone

seppestas, 2017-03-04:
I seem to have a similar problem on my Fabtotum Core edition with original Raspberry Pi 1 using a class 10 32GB Sandisk SD card formatted using MacOS's disk utility.

Disk utility did not seem to find any new the partitions on the SD card, only the shrunken FAT partition.

Here is a picture of the screen attached to the RPi:
[image]

seppestas, 2017-03-05:
Same result with a 128GB class 10 Lexar SD card. I'm now looking for my 8GB SD card I should have laying around somwhere, but I mostly own >32GB SD cards #firstworldproblems