FABtotum/fabui-colibri #21: First setup button doesn't work

bilderbuchi, 2017-02-26:
When I go to dashboard-maintenance-first setup, I get "Error requesting maintenance/first-setup: 500 Internal Server Error".
I just freshly installed Colibri, all other menu items so far are accessible, I was affected by #11 afaict, but now after a reboot I am on a fully updated machine.
```
OS 	Linux fabtotum 4.4.38-v6 #1 Thu Feb 23 14:02:32 CET 2017 armv6l GNU/Linux
Firmware 	v1.0.0097.1
FabUI 	v0.9
Hardware 	v1
Installed Head 	hybrid_head
```

LuxAetern, 2017-02-27:
+1 
have the same error with the exact same config

krios-fabteam, 2017-02-27:
Hi guys, 
we'll add again the feature before the official release

daniel-fabteam, 2017-04-11:
First-setup has been returned with a new look.

closed 2017-04-11 by daniel-fabteam

bilderbuchi, 2017-04-12:
thanks! I hope I'll try it out soon!