FABtotum/fabui-colibri #22: Shutdown issues

NeoBelerophon, 2017-02-27:
Hi I found the flowing issues while shutting down:
- shutdown dialog shows "Hello undefined"
- shutdown progress in the GUI runs for ever? (at least 20 min.) 
( after I was unable to log in via ssh I turned it off)

It's installed on an RPi3 and Totumduino 2

daniel-fabteam, 2017-02-27:
Hi Neo,

We have just released a new sdcard image. The issue should be fixed.
Could you please try the new image and see if it fixed the issue?

Also, after you hear the beep sound when you click on shutdown. 5 seconds after that RPi is in a safe state and the unit can be switched off. There is no automatic power supply off switch.

bilderbuchi, 2017-02-28:
> Also, after you hear the beep sound when you click on shutdown. 5 seconds after that RPi is in a safe state and the unit can be switched off. There is no automatic power supply off switch.

Now *that* is some useful information that really should be in the shutdown dialog!

daniel-fabteam, 2017-02-28:
The shutdown issue and "undefined" username should be fixed in the latest update.
Also check out the new [guide](https://github.com/FABtotum/fabui-colibri/wiki/SDCard-formatting) for how to format the sdcard.

@bilderbuchi Is a good idea. We will put a comment on the shutdown popup.

closed 2017-02-28 by daniel-fabteam

bilderbuchi, 2017-02-28:
Name is correct on the shutdown dialog now. no comment yet, should I open an issue to track that separately?