FABtotum/fabui-colibri #24: No Internet after configuring the ethernet

dr4Ke, 2017-02-27:
I had trouble configuring the wifi with the previous sdcard image, so I played safe this time, and configured only the ethernet with DHCP on my home network.
I ended up with a valid IP address, DNS resolution, but no Internet access. The gateway was unreachable.
The cause was the routing table:

    [root@fabtotum ~] netstat -rn
    Kernel IP routing table
    Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
    0.0.0.0         172.16.0.1      0.0.0.0         UG        0 0          0 wlan0
    0.0.0.0         192.168.0.250   0.0.0.0         UG        0 0          0 eth0
    172.16.0.0      0.0.0.0         255.255.0.0     U         0 0          0 wlan0
    192.168.0.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0

The default gateway was set twice, and the first one was the wifi, unconfigured connection.

The solution, for me, was to set up the wlan interface as well, but I wonder what would happen for someone without wlan.

daniel-fabteam, 2017-02-27:
@dr4Ke Could do the following experiment?

> cd /etc/network/interfaces.d
> cp wlan1 wlan0
> sed -i wlan0 -e "s/wlan2/wlan0/g;/gateway/d"
> reboot

daniel-fabteam, 2017-02-28:
There is a new sdcard which should fix the issue with gateways.

closed 2017-02-28 by daniel-fabteam

dr4Ke, 2017-02-28:
I confirm that this issue is fixed for me.