FABtotum/fabui-colibri #26: OS always corrupted after reboot/shutdown

florentmaye, 2017-02-28:
Each time I stop/reboot my Fabtotum, the stopping process never end (I waited more than 1 hour!!) and I have to manually shut it down. Then when restarting it, the OS show at least a recovering screen (and recovering lead to the same problem) but most of time it just doesn't even restart. It is the fifth time I tried the OS, I'm wondering what's wrong because it seem a so obvious issue that evrybody should have it ...

Maybe putting the OS on a usb stick could resolve that ...

daniel-fabteam, 2017-02-28:
Could you try to follow the new [sdcard formatting](https://github.com/FABtotum/fabui-colibri/wiki/SDCard-formatting) guide just to make sure is not related to sdcard partition misalignment? Also make sure to download the sdcard again as we have release a new one today.

When you click on shutdown, 5 seconds after the beep the system is in safe state and can be turned off.
There is no automatic power off hardware support so the unit will never switch off on it's own. You still need to turn the switch off like on the previous FABUI.

BramElderman, 2017-02-28:
Since the original ticked was gone i will poste my findings here,.. (no p option avalable)

Faulty Card:
Brams-MacBook-Pro:~ bram$ sudo fdisk /dev/rdisk2
Disk: /dev/rdisk2	geometry: 960/128/63 [7741440 sectors]
Signature: 0xAA55
         Starting       Ending
 #: id  cyl  hd sec -  cyl  hd sec [     start -       size]
------------------------------------------------------------------------
 1: 0B 1023 254  63 -   65 197   5 [      8192 -    1048449] Win95 FAT-32
 2: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
 3: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
 4: 00    0   0   0 -    0   0   0 [         0 -          0] unused      

Good working Card: (big difference explains a lot i think)
Brams-MacBook-Pro:~ bram$ sudo fdisk /dev/rdisk2
Password:
Disk: /dev/rdisk2	geometry: 951/255/63 [15278080 sectors]
Signature: 0xAA55
         Starting       Ending
 #: id  cyl  hd sec -  cyl  hd sec [     start -       size]
------------------------------------------------------------------------
 1: 0B 1023 254  63 -   65  73   5 [         2 -    1048827] Win95 FAT-32
 2: 83  791  38   5 -  315  37   1 [   1050624 -    2097152] Linux files*
 3: 83  315  37   2 -   77  36   2 [   3147776 -    1048576] Linux files*
 4: 05   77  36   3 -  332   7   5 [   4196352 -   11081728] Extended DOS
Signature: 0xAA55
         Starting       Ending
 #: id  cyl  hd sec -  cyl  hd sec [     start -       size]
------------------------------------------------------------------------
 1: 83   83   2   1 -  869   1   1 [   4198400 -    1048576] Linux files*
 2: 05  643   3   1 - 1023 254  32 [   5246976 -   10031104] Extended DOS
 3: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
 4: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
Signature: 0xAA55
         Starting       Ending
 #: id  cyl  hd sec -  cyl  hd sec [     start -       size]
------------------------------------------------------------------------
 1: 83  874  40   5 -  332   7   5 [   5249024 -   10029056] Linux files*
 2: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
 3: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
 4: 00    0   0   0 -    0   0   0 [         0 -          0] unused      
Brams-MacBook-Pro:~ bram$

BramElderman, 2017-02-28:
oops pasting it made the formatting bizar

daniel-fabteam, 2017-02-28:
@BramElderman The first sdcard clearly did not make it through the partitioning process. 
For the working one is kind of a strange representation but make sense. First 4 partitions (max that is supported by MBR partition table) are there. One FAT32, two linux partitions and an extended one that holds two more linux partitions which are **changes** and **userdata**.

@BramElderman Which procedure in the sdcard formatting guide did you use to make the working sdcard?

PS.: Use *insert code* when you want to prevent the format to go crazy. Also here on github you can switch to *Preview* tab before you post the comment to see how it looks.

florentmaye, 2017-02-28:
Sorry, I didn't see previous posts. Problem resolved.

Thank you

closed 2017-03-01 by krios-fabteam

florentmaye, 2017-03-02:
I can now reboot my fabtotum, but at every boot I have a recovery screen. I recover then it work fine, but next but => recovery screen ...

daniel-fabteam, 2017-03-02:
What kind of head are you using?

If the head is not present during startup it will trigger the recovery mode.

florentmaye, 2017-03-02:
Printing v2

florentmaye, 2017-03-02:
I restarted it again, and this time no more recovery screen (I remove the head for cleaning and put it back), it worked perfectly this time ...