FABtotum/fabui-colibri #30: [improvement] scan visualization

florentmaye, 2017-02-28:
Currently when I perform a scan (I tested laser), I can only see the raw file. It would be nice to have a tool to visualize them, and even convert them in appropriate format.

bilderbuchi, 2017-02-28:
Also, when I finished a laser scan yesterday, the window just closed when I wasn't looking, and went back to the scan method selection, with no indication of completion - later on I found the created file in the project manager by accident. :-/

daniel-fabteam, 2017-02-28:
@bilderbuchi  Could you report the described issue as a separate one as this one was flagged as **feature request** Also specify which type of scan you used, will make it easier to track the bug.

bilderbuchi, 2017-03-01:
Sure thing: #36.