FABtotum/fabui-colibri #32: Jog arrows don't work

florentmaye, 2017-02-28:
The jog arrows in the Jog panel or in shortcut on top right doesn't do anything when I send commands, even after sending head to home

daniel-fabteam, 2017-03-01:
Could you test the jog panel again but make sure not to use raspicam arrow buttons before. Or simply try if it works when you reload the page with F5.

daniel-fabteam, 2017-03-02:
Should be resolved in the latest update. If it still happens, please re-open this issue.

florentmaye, 2017-03-02:
Nope, still doesn't work in any mode (I'm using chrome btw)

florentmaye, 2017-03-02:
doesn't work even with firefox

daniel-fabteam, 2017-03-02:
Chrome and Firefox on Windows, Mac or Linux?

Make sure you repeat your steps and get the jog to stop working. Then...

There is a plugin called **System logs**.  Go to `Plugins -> Online` and install it. Once installed go to `Plugins -> Installed` and activate it.
With this you should get a new menu item called `SYSTEM LOGS`. Go there and and click on *Download*. Wait a little bit until the logs are compressed and a download should start. Attache the resulting `archive.tar.gz` here.

florentmaye, 2017-03-02:
On Windows.

It doesn't stop working, it never worked.

[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/814937/archive.tar.gz)

daniel-fabteam, 2017-03-02:
Go to jog view and press one of the arrow keys buttons. Copy and paste the output from the command history (bottom right list box) here. There should be something like

    G91
    ok
    G0 ...
    ok
    M400
    ok

Next check if the head will move if you type in the `WRITE COMMAND` section

    G91
    G0 X+10.00 F1000.00
    
Use `SHIFT+ENTER` if you are typing the commands or un-check "Press **ENTER** to". Otherwise the command will be send as soon as you hit `ENTER`.

Or just copy paste the commands above and press **Send**.

daniel-fabteam, 2017-03-02:
Do the small experiment I describe above because I found that the jog control sends **GO** instead **G0**  (G "Zero")
Just to confirm that is related to this.

This is probably what is causing this issue. Not sure how it worked here anyway. The fix will be available in the update tomorrow.

florentmaye, 2017-03-02:
> press one of the arrow keys buttons
This doesn't do anything for me. Jog buttons don't work.

The head moved and outputs:

```
G91
ok
G0 X+10.00 F1000.00
ok
```

And I did the same test I did before and this time it worked, I don't know why (because I copied pasted command from results, I shouldn't have made any error) ... Ang I think I have a G0 not a GO...

daniel-fabteam, 2017-03-03:
Check if the issue is fixed in today's update.

florentmaye, 2017-03-03:
Nope still don't work

daniel-fabteam, 2017-03-06:
We have done more safety checks in today's update. Could you make a fresh sdcard and try it the jog is working?

florentmaye, 2017-03-06:
I did the last update and it still doesn't work. It is like if buttons where never clicked.

> Could you make a fresh sdcard and try it the jog is working?

Is it different than doing the update throw the normal update mechanism?

daniel-fabteam, 2017-03-06:
If you system is in a normal condition the update should do the trick but if there are some altered configuration files then a fresh sdcard is better. At least during the beta testing phase.

daniel-fabteam, 2017-03-07:
Let's do the complete check again to try and find out where it's failing.

* Open Chrome
* I the menu go to *More Tools* -> *Development Tools*
* Go to *Network* tab and click on *Disable cache*
* Hit `F5` to refresh the page
* Switch to *Console* tab. There should be a lot of messages
* Make sure you have the **System logs** plugin installed
* Go to Jog view and click on every arrow two times with a small pause between clicks (just to be sure)
* In the *Console* tab Right-Click and select *Save as...* and attach the log here
* Go to **System logs**, click *Download* and attach the `archive.tar.gz` here

daniel-fabteam, 2017-03-07:
Also try do the following.

In Jog command type `M18` and move the head to the center.
After that run `M119` and paste the output here.

christopherwit, 2017-03-17:
With latest update in IE11 jog seems to work ok (accept for the estetic of the buttons staying grey after mouse has hoovered once over a button).
Nevertheless in firefox macros are prefectly operational, but jog buttons aren't.

florentmaye, 2017-03-19:
Sorry for the long silence.
I have reflash my sd card with the last version, and it still don't work. It is like if the buttons weren't clicked at all. (no visual feedbacks, no commands sent ...)

daniel-fabteam, 2017-03-20:
As it has been a long time. Which browser are you using?

florentmaye, 2017-03-20:
Chrome and I tested too with Firefox with no result

christopherwit, 2017-03-22:
I just noticed jogpanel doesn't work in firefox and chrome when using my touchpad or mouse, but when I tap my touchscreen it does work correctly in both browsers.
IE11 all is good.

PaoloP74, 2017-03-24:
After pressing the home button and the print head is in the lower left, by pressing the down arrow, the red pointer scrolls down the page.
Fabui 0.99.20170323.1742 Build date: 2017-03-23

christopherwit, 2017-03-26:
In latest release 0.99.20170323.1742 a new issue is introduced for firefox for the top page jog and temperature dropdowns. When I try to open them they get immediately closed again.
In IE11 dropdowns appear and all buttons are functional but stay masked grey after mouse over action.
In Chrome dropdowns appear correctly but head doesn't respond to xyz buttons. Temperature control is fine.

On the jog page, none of the xyz buttons make the head/bed move around in Chrome and firefox.
Crosshair functionality is OK after xy homing.
Again in IE11 all buttons are functional but stay masked after mouse over.

christopherwit, 2017-03-26:
well the dropdowns are not always closed immediately. Not sure how to replicate this behavior. Sorry. When I have a more specific cause I'll let you know.

daniel-fabteam, 2017-03-27:
When trying to replicate the error in chrome/firefox open the javascript console using `CTRL+SHIFT+I` and switch to `console` tab.

Once you manage to replicate it save the console log in to a file and attache it here. Also attach the system logs that you can download once you install system logs plugin.

krios-fabteam, 2017-03-30:
Should be fixed in the latest updates. If it still happens, please re-open this issue

christopherwit, 2017-03-31:
Sorry, still no xyz response on the jog page when browsing with Firefox 52.0.2 32 bits

daniel-fabteam, 2017-04-03:
When trying to replicate the error in chrome/firefox open the javascript console using CTRL+SHIFT+I and switch to console tab.

Once you manage to replicate it save the console log in to a file and attache it here. Also attach the system logs that you can download once you install system logs plugin.

christopherwit, 2017-04-10:
Hi guys,

Is there any progress on the functionallity of the jog buttons(both from the dropdown panels and the jog page) in firefox?

Dropdown panels aren't closed anymore when opening them. But the complete jog functionality as in a move response when pressing xyz buttons is still broken in Firefox.

I checked the console output when clicking the xyz buttons, but no new lines other than temperature updates pop up.
When using jog buttons no jquery line comes by similar to when I use the home button or crosshair on the bed view on the jog page.

daniel-fabteam, 2017-04-11:
Could you also add the full javascript console log (`CTRL+SHIFT+I` )?

It might seem that there is noting strange in the log but please do add it anyway.

christopherwit, 2017-04-11:
No problem, I will reproduce the issue this evening and send you (all) new
logs.

christopherwit, 2017-04-11:
Here you go;
[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/913937/archive.tar.gz)
[java log second visit jog page.txt](https://github.com/FABtotum/fabui-colibri/files/913935/java.log.second.visit.jog.page.txt)
[java log.txt](https://github.com/FABtotum/fabui-colibri/files/913936/java.log.txt)

I posted two java logs because I noticed some more messages passing by when I loaded the jog page for a second time. Hope this information helps in solving the issue.

christopherwit, 2017-04-18:
Did the logs provide you any usefull info to get this longlasting issue solved?

christopherwit, 2017-06-10:
Dear fabteam, this issue isn't still fixed! Therefore don't close this issue! Please reopen. XYZ axis only move when using touchscreen input on jog page and dropdown panel in Chrome and Firefox. When using regular usb mouse or touchpad to tab the jog buttons nothing happens.

By using touchscreen input I do get a response on the serial monitor(bottom right) but with a regular mouse nothing appears. Hence why there is also nothing happing on the javascript console.
Please fix before closing issue tracker.

christopherwit, 2017-06-10:
Ok after some more digging Firefox and Chrome seem to be missing a mousedown evenlistener.

christopherwit, 2017-06-14:
please reopen

christopherwit, 2017-06-14:
Hi daniel, Thanks for reopening this issue. but I just came across the
official announcement of the new full release of fabui colibri 1.0. As the
update isn't working because updateserver is currently down, I will have to
wait and see if the update fixed this issue.

daniel-fabteam, 2017-06-14:
As we have released the full release yesterday you should download a new sdcard image and do a clean flash and system install including all calibrations.

Seems that chrome did change some event names so I will take a look into it and make adjustments so the codes adapt. Will inform you when the update will contain the fixes.

christopherwit, 2017-06-14:
Please be aware the issue also applies to Firefox, but I suspect the two
are related to the same cause.

daniel-fabteam, 2017-06-22:
@christopherwit Does the jog touch control work?

christopherwit, 2017-06-28:
Hi sorry I forgot about your mail. No it still doesn't work in Firefox.

PaoloP74, 2017-12-06:
When the head is in the bottom right corner, looking at the printer at the front, pressing the diagonal arrow of the jog towards the corner the endostops do not work and the motors start. Pressing the up or right arrow the endostops work properly and the head remains still.
I have not tried the other corners and the other arrows.
FabUI 1.1.0

closed 2018-04-04 by krios-fabteam

florentmaye, 2018-04-04:
I confirm you tonight but last time I tried with the mouse, it didn't work,
only with touch...

krios-fabteam, 2018-04-04:
Please, could you report also fabui version, browser name and version and operating system?

christopherwit, 2018-04-04:
Sorry guys with latest update only touch is working in both firefox and
chrome. Mouse clicking is still not working. For both click and touch cases
pseudo clases(#184) are now correct but button only gets active when I
click inside the button area but not on de icons themself (arrows etc). In
case of tapping the location of contact doesn't matter.

Version info per your request:
Core 1.0.3, kernel 1.0.0, web 1.0.0, Libs 1.0.1, python 1.0.0, python extra
1.0.1 network 1.0.4, network extra 1.0.0, scientific 1.0.1, tools 1.0.1,
rpi 1.0.0, samples 1.0.1, fabui 1.1.2
Chrome build 65.0.3325.181 64bits
Firefox 59.0.2 64bits

krios-fabteam, 2018-04-04:
I'm using the same browsers with the same versions but I can use the jog without any kind of problems :(
I'm accessing fabui using a pc with Windows 10 as the operating system, and you?

I see that your fabui version is 1.1.2, yestarday we released an update, even it will not fix your issue please update so we are aligned

christopherwit, 2018-04-04:
Hi Krios, I was just about to write an update since my interface just
notified me on new fabtotum updates:)
I did the complete update(fabui, network, fablin etc) including reboot, but
issue remains the same. Let me know if you want teamviewer access or
something similar to check stuff out on my system. Let me know how to share
details if needed.

Yes I'm running windows 10 home 64 bit version 1709 build 16299.309

krios-fabteam, 2018-04-04:
A TeamViewer session is a good idea, send the details to km@fabtotum.com so I can enter and check

krios-fabteam, 2018-04-04:
The issue is present on laptops with touch screens, so browsers get messed up with the click events, who his having this issue should disable "Touch events" in the browser settings

florentmaye, 2018-04-04:
I confirm I also have a touch screen.

christopherwit, 2018-04-04:
To refine Krios's remark. For those experiencing this issue in chrome;
navigate to chrome://flags/#touch-events and make sure to set touch events
API to disabled.

For Firefox users I've not found a workaround yet.