FABtotum/fabui-colibri #34: PhP Error

florentmaye, 2017-02-28:
I was in the jog menu performing operations, then I open the door to make something, and when I came back:

```
A PHP Error was encountered
Severity: Warning

Message: implode(): Invalid arguments passed

Filename: libraries/JogFactory.php

Line Number: 93

Backtrace:

File: /usr/share/fabui/application/libraries/JogFactory.php
Line: 93
Function: implode

File: /usr/share/fabui/application/helpers/fabtotum_helper.php
Line: 496
Function: sendCommands

File: /usr/share/fabui/application/helpers/fabtotum_helper.php
Line: 299
Function: doGCode

File: /usr/share/fabui/application/controllers/Jog.php
Line: 26
Function: isHeadInPlace

File: /usr/share/fabui/index.php
Line: 316
Function: require_once
```

```
A PHP Error was encountered
Severity: Warning

Message: implode(): Invalid arguments passed

Filename: libraries/JogFactory.php

Line Number: 93

Backtrace:

File: /usr/share/fabui/application/libraries/JogFactory.php
Line: 93
Function: implode

File: /usr/share/fabui/application/helpers/fabtotum_helper.php
Line: 496
Function: sendCommands

File: /usr/share/fabui/application/helpers/fabtotum_helper.php
Line: 315
Function: doGCode

File: /usr/share/fabui/application/controllers/Jog.php
Line: 27
Function: isBedInPlace

File: /usr/share/fabui/index.php
Line: 316
Function: require_once
```

krios-fabteam, 2017-03-01:
[image] [PhP Error](https://trello.com/c/HcOWinIE/180-php-error)

florentmaye, 2017-03-01:
What do you mean? I don't think I did something special out of the normal
use of the UI  ...

krios-fabteam, 2017-03-01:
@florentmaye sorry, that mail wasn't for you

krios-fabteam, 2017-03-01:
Fixed in the latest FABUI update.
Please confirm if it really was fixed.

closed 2017-03-01 by krios-fabteam