FABtotum/fabui-colibri #37: Fresh install is corrupt right after the first update

IT2BE, 2017-03-01:
Installed Colobri on a 2nd SD Card.
Started FABtotum.
Everything went well.
Moved to Wifi.
Received notifications for update.
Updated.
Saw that updated were applied.
System Freezed.
Restarted.
Nothing :(

daniel-fabteam, 2017-03-01:
Hi,

Which version of FABtotum do you have? RPi1 or RPi3 based?
Also did you make a new sdcard from yesterdays image?

There have been some changes in the update process so the early sdcard images might not work.
We release a new sdcard image every day. The one from yesterday might work better.

daniel-fabteam, 2017-03-01:
Another question, which operating system did you use to format the sdcard.
We have a guide on how to do it as there were issues for fresh installs after the first shutdown/reboot.
So let's make sure is not any of those scenarios.

daniel-fabteam, 2017-03-03:
Has this issue been resolved with the new sdcard image?

closed 2017-03-06 by daniel-fabteam

IT2BE, 2017-03-06:
I never received any notification until this was closed :(
I have an RPI3 based FABtotum (upgraded).
The SD Card was freshly made using the instructions given on your website.

I have not yet taken the time to create a new image yet...

daniel-fabteam, 2017-03-06:
When you create a new one check with a partition manager that there is no unalocated space before the initial FAT32 partition.

Also before doing and update with a fresh system. Just restart it first and see if it comes back or if it fails.
The sdcard formatting issue is quite annoying one but when done right everything should work.

Make sure to download the latest image as we released a new one for today.