FABtotum/fabui-colibri #38: localization

homahe, 2017-03-01:
After having installed colibri successfully, changing the interface language to German, the texts in I/F  show, that a localization didn't take place for the character-set used.

Trying to change that in the cli as root, _locale_ appeared not to be a command, neither _localectl_ exists.
Changing the default entry in locale.conf to de_DE.UTF-8 seems not to work after reboot, and /etc/vconsole.conf doesn't exist...

IMHO when the U/I is changed during set-up fo colibri or even later in the user-preferences LANG and KEYMAP should be changed accordingly .

daniel-fabteam, 2017-03-01:
Hi,

The system localization does not change and is always set to en_US.UTF-8. Only the UI part does adapt and all the calls to applications in the background change to for example de_DE.UTF-8.

By "the texts in I/F show" what do you mean? Which view does not show localization? Network?

Translations are still under development so there will be missing text in German and will fallback to English.

homahe, 2017-03-01:
Hello,
what is meant by "the texts in the I/F show" is, that for the German 'umlauts', ae-lig, Ae-lig etc. a wrong character is shown. 
The second problem is, that when there is no possibility to adjust LANG and KEYMAP to locale poor root has either to remember a US keyboard-layout or to attach such :-(

daniel-fabteam, 2017-03-01:
Switching localization for the whole system could have unexpected side-effects. However only allowing the keymap to be adjusted should not affect system operations.

At this point we are focused to fix blocking issues and will address extra features at a later point.

It would help a lot if you could check the German translation in the mean time. 

For now don't change any system settings manually as it could make future updates misbehave and create issues that will be only specific to your system.

If there is a scenario in which something is not working create an issue that will help us reproduce it.
Feature requests are to be avoided during beta testing phase as it's stated in the Issue reporting rules.