FABtotum/fabui-colibri #39: Network configuration

homahe, 2017-03-01:
Changing the #Network-settings in /earlyboot/earlyboot.conf  to the necessary settings to access the device in a local network are clobbered by the predefined addresses during the set-up instead of copying them. 
It would be much easier for the user to modify the entries in earlyboot.conf in many cases than to adjust a pc used to access the FABTotum-printer and a maybe restricted router to the fixed 169.254.1/16 network.

During the set-up in many cases the download will fail, because the gateway address cannot be changed.

As the earlyboot.conf is accessible easily before the set-up is started using a standard editor IMHO adjusting the network-configuration would be much easier this way. One could even consider to have this part of the earlyboot.conf as a separate file.

daniel-fabteam, 2017-03-01:
Is your issue that you cannot use 169.254.1.2 in your network?

`earlyboot.conf` is not intended to be modified. So please explain your issue so we can try to solve it.

homahe, 2017-03-01:
Hello, Daniel,

sorry I couldn't make it clear:

Sitting behind a gateway/router which will not allow to modify its
address, here e.g. 192.168.99.1, the whole network is configured as a
192.168.99.0/24 network.

So to access the FABTotum one has to change a PC to the fixed
169.254.1.0/16 or /24.
When the set-up is running, it will not be able to acces the gateway, huh?

If one changes the addresses in the ' not intended to be modified
earlyboot.conf ' you will be able to start the setup via the web-I/F.
But during the set-up stubbornly the entries in the earlyboot.conf are
ignored. It would cause no harm to accept the network-entries, would it?

So, after the first step of the installation one has no access to the
FABtotum anymore, but using the root cli one can change the
network-parameters again to adjust them to the actual environment. And
voilà, Fabtotum can be reached, the set-up continues peacefully reaching
out via the gateway to the internet.

So, YES, I cannot use 169.254.1.1 as a gateway and 169.254.1.2.

And it hasn't been a problem.

Maybe others, who cannot use 169.254.1/16 too, will have a problem. But
editing a simple file so it represents the network necessities  before
starting the installation most of 'em will be able to. Would be easy to
write an explanation of what to do there...

Just my humble opinion. And you asked your users how to improve Colibri,
didn't you?

Best regards

M. E. Holzé

daniel-fabteam, 2017-03-01:
Hello again,

Now I understand the problem. Is true that earlyboot.conf network setting is not transferred to the installed system as we never considered that scenario. We will make it configurable and put a description how to tune it on the wiki at a later stage. Before or maybe after full release.

Vielen Dank,
Daniel

christopherwit, 2017-03-12:
Please make sure this issue is also taken into account for autorecovery. It would be really beneficial if the altered setup is also used during recovery.
Thanks if it already is, as I did not tried the autorecovery yet.

daniel-fabteam, 2017-03-12:
This feature will be implemented at a later point and be documented in the wiki.

daniel-fabteam, 2017-03-31:
Is available in `core-0.6.5` bundle update.

closed 2017-03-31 by daniel-fabteam

christopherwit, 2017-03-31:
Great job guys! Working out of the box. (I do know my way around earlyboot file, but please don't forget to include instructions close to the image download link).