FABtotum/fabui-colibri #43: After Engraving Laser Head does not return to image orgin

AndrewLShaw, 2017-03-01:
It's not possible to repeat a burn over the same engraving because the head does not automatically return to the image origin.

Also there is no option to use the stored calibrated settings for the Z- height.

BramElderman, 2017-03-03:
How is it possible that you have laser option at all...??
is the plugin available??

daniel-fabteam, 2017-03-03:
The plugin has been uploaded but the rest is not ready yet. As soon as we are ready will release the update for today. It will include fixes for the laser and allow the plugin to work.

AndrewLShaw, 2017-03-03:
@BramElderman once you 'Install' the laser head you're able to use the laser functionality. It's not as well honed as with current release but it definitely works.

daniel-fabteam, 2017-03-03:
We have released a new update. Laser functionality is moved to a laser plugin.
You will need to download it from the `Plugin -> Online` section and activate it. 
There should be a new menu item in `Create` and Projectmanager will recognize laser gcode and show **Engrave** action next to them.

closed 2017-03-03 by daniel-fabteam

BramElderman, 2017-03-03:
Ah thanks Andrew,.. reinstalled it so many times from scratch that I forgot that I also replaced the head back to the hybrid... retried and YES the functionality is back indeed and I can install the plugin once the head is in place.