FABtotum/fabui-colibri #45: No bed calibration

homahe, 2017-03-01:
With actual FABUI 0.99.20170301.1711
when choosing bed calibration, the introduction screen is shown, hitting the start button is rewarded with two faint high pitch beeps, nothing moves at all, esp. the bed does not, the head doesn't home and the probe isn't switched out. But after a while one is rewarded with a 100% level bed. 
This behavior doesn't change after reboot or reset.

florentmaye, 2017-03-02:
Same for me. I'm sad I think my bed leveling is not right anymore ...

homahe, 2017-03-02:
One thing to mention:

In the area below the 'Head' temperature slider etc., right part of the screen, was mentioned 'No bed installed' (as far as I remember, I'm back to the old version I confess..), but the bed was installed all right and the temperature reading was working +/- 1 degree correctly (measured by IR scanner). 
When checked with the old pgm, it was 100% level. So I'm quite sure it's an unwanted feature ;-)
BTW.:
Nozzle height calibration: 
I got an error, reading: MaxHeight None Type has not attribute '__getitem__'

The problem with the feeder calibration has already been mentioned; got the same too

daniel-fabteam, 2017-03-02:
@homahe  "Bed not installed" confirmed and created as a new issue #51 to track it.

daniel-fabteam, 2017-03-03:
There is a plugin called **System logs** in the `Plugin->Online` section. Install it and activate it.
Repeat the steps you made to reproduce the bed calibration issue then go to **SYSTEM LOGS**, press download and attach the resulting `archive.tar.gz` here.

krios-fabteam, 2017-03-03:
ixed in the latest FABUI update.

closed 2017-03-03 by krios-fabteam