FABtotum/fabui-colibri #46: Make menu has no LASER item in the list

BramElderman, 2017-03-01:
I see people discussing the laser engraving however i don't have the laser option available in the menu,... is that because i have the RPi 1?? the old software has it available and i have done laser engraving with that... in the Colibri release it is nowhere to be seen 

i only have **print mill scan and history** in the make menu

[image]

Maintenance System info also displays weird data,.. for the record i removed the laser head and reinstalled the hybrid head.

[image]

daniel-fabteam, 2017-03-02:
The laser menu item is disabled for now as it's still not fully ported to fabui-colibri. It will be available as a plugin which will serve also as an advanced example of how to make plugins in the new framework.

For the System Info issue. I had this few days ago but was resolved. If after refreshing the page it's still there open an issue so we can track it. That way if someone else has the same problem they can join the discussion.

daniel-fabteam, 2017-03-03:
The latest update is released. Laser menu item is removed as well as all the logic behind it from main fabui.

However now we have a plugin to add laser functionality for people who need it.
Check it out and report any issues you find.
Is not completed yet but should allow you have some fun with the laser.

`Note: the upper folder icon in jog view is "Restore XY", the lower folder icon is "Restore Z". If you home XYZ first, when you press "Zero" the position will be store and can be restore next time you do some laser engraving`

closed 2017-03-03 by daniel-fabteam