FABtotum/fabui-colibri #47: command console doesn't work

florentmaye, 2017-03-02:
In jog mode, I tried to send commands using the console but it didn't work. I even tried to copy paste these generated when clicking to move the head (only interaction  which work in jog mode with homing) but nothing.

florentmaye, 2017-03-02:
Well in fact since I did that no other command work (even homing doesn't respond ...)

daniel-fabteam, 2017-03-02:
Is your unit up-to-date?

florentmaye, 2017-03-02:
Yes with the last version "Great! Your FABtotum Personal Fabricator is up to date"

daniel-fabteam, 2017-03-02:
When you type `M503` do you get a response?

florentmaye, 2017-03-02:
```
M503
FABtotum TOTUMDUINO
 Batch Number: 5
FABlin
 Version: V 1.0.0097.1
 Baudrate: 250000
Steps per unit:
  M92 X72.58 Y72.58 Z2133.33 E1554.64
Maximum feedrates (mm/s):
  M203 X550.00 Y550.00 Z15.00 E23.00
Maximum Acceleration (mm/s2):
  M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
  M204 S4000.00 T100.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
  M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm):
  M206 X0.00 Y0.00 Z0.00
PID settings:
   M301 P20.00 I3.50 D30.00
Servo Endstop settings: R: 31 E: 131
Z Probe Length: -31.55
Installed head ID:
 M793 S2
ok
```

daniel-fabteam, 2017-03-02:
Command line seems to work. The issue is with the jog controls.

closed 2017-03-02 by daniel-fabteam

florentmaye, 2017-03-02:
Possible, however it is strange that my test to find a way to jog without the buttons failed ... I may have done a mistake but I'm surprised since as I said I copy past commands.