FABtotum/fabui-colibri #56: [improvement] possibility to disable a network interface

dr4Ke, 2017-03-04:
I think it would be better if we had the possibility to just disable the ethernet or wifi interface. You could add a `Disable` choice in the interface together with `Static` and `Automatic (DHCP)`. In this case, the sane default for WiFi would be `Disable`.

daniel-fabteam, 2017-03-04:
What would be the benefit of doing so?
By default ethernet and wifi are configured to static values without gateway. They don't take any significant time to setup during boot in that state.

dr4Ke, 2017-03-04:
I guess it's just more satisfying. Plus, I wonder what would happen if one is using an ethernet network in the same range than the predefined WiFi static address.
Well, I tried this, and… it's working fine.

BramElderman, 2017-03-04:
in a sense you already have that, if you don't insert the wifi stick the wifi tab is not available,.. it is ethernet only and works fine for me,.. only if you use both it might become confusing

christopherwit, 2017-03-12:
In my opinion a button to disable wifi is a nice to have.

It does get a bit confusing when you run from a pi3 which has the wifi enabled by default, and both interfaces get utilised at the same time. Especially when fabui reports no internet connection but is connected to your network.

daniel-fabteam, 2017-03-14:
Disable option for wireless interface is available in the latest update.

closed 2017-03-14 by daniel-fabteam