FABtotum/fabui-colibri #58: M763 aborted log messages

dr4Ke, 2017-03-04:
The FabtotumServices.log is flooded with these messages:

```
DEBUG : put on command queue: M763,bootstrap
DEBUG : << M763 [RQ: 0]
DEBUG : Waiting (3) for [M763,bootstrap] aborted: False
DEBUG : Waiting (3) for [M763,bootstrap] aborted: False
DEBUG : Waiting (3) for [M763,bootstrap] aborted: False
DEBUG : Waiting (3) for [M763,bootstrap] aborted: False
DEBUG : Waiting (3) for [M763,bootstrap] aborted: False
DEBUG : Waiting (3) for [M763,bootstrap] aborted: False
```

There is more than 1000 messages like this after only one hour uptime.

dr4Ke, 2017-03-04:
I think the arduino card is not responding. I can't update its firmware, and the UI is incapable to show me the current version as well… I fear it's dead.

daniel-fabteam, 2017-03-04:
Try to use the **Emergency** button. This will restart both the totumduino and FabtotumServices.

If that is not working you can try to do it manually over ssh executing
`/etc/init.d/fabui emergency`

In any case it is safe to poweroff the unit even if it's in this state and turn it on again.

You can flash the totumduino even if it's firmware is corrupted. Ignore the missing version info and select 1.0.0097.1 firmware and flash.

dr4Ke, 2017-03-04:
Thank you for your reply. In the meantime, I reverted to the previous version of fabui (not colibri), reset the controller, and did the fabui update. The controller is back on. I'll retry the colibri version now, and see if the problem is gone.

dr4Ke, 2017-03-04:
OK, it's fine now. It was the controller not responding. Sorry for the false alert, and thank you for the insight.

closed 2017-03-04 by dr4Ke