FABtotum/fabui-colibri #60: head blocked

florentmaye, 2017-03-04:
At the head of a print, the head is often at top right corner. 
If I want to move it, I'm using the home button in jog mode, however it seem this process try to go over printer limits and the following warning message is shown:

>   X min Endstop hit: Move the carriage to the center or check Settings > Hardware > Custom Settings >Invert X Endstop Logic
 Press OK to continue or Ignore to disable this warning

daniel-fabteam, 2017-03-04:
If I remember correctly. You should have a v5 unit. It has inverted endstop logic.
Go to `Settings -> Hardware` and see if you are using "Default settings" or "Custom settings".

Select *Custom settings* to get more options. The *Invert X Endstop Logic* if you have a v5 unit should be set to **YES**

daniel-fabteam, 2017-03-06:
To make sure all settings are at factory settings you can run `M502` in the jog command. This will restore firmware settings to their defaults in case some of them have been store wrong.
After this restart your unit.

daniel-fabteam, 2017-03-16:
Is this issue still present with the latest sdcard image?

RobertsVancovics, 2017-03-16:
I had the same message. Inverting the x endstop logic solved it.

closed 2017-03-27 by daniel-fabteam