FABtotum/fabui-colibri #61: Partition problem at setup time

UnguidedFreedom, 2017-03-04:
I initially formatted my 8 GB SD card (the one given with my Fabtotum) as FAT32 labeled "BOOT" and extracted the zip on it.

It booted successfully, so I started the installation. During that process, I had an external monitor hooked up to the Raspberry Pi, and was therefore able to see the following errors:

```
mount: mounting /dev/mmcblk0p5 on /mnt/bigtemp failed: No such file or directory
mount: mounting /dev/mmcblk0p6 on /mnt/userdata failed: No such file or directory
```

[image]

Once the installation was over, I had access to the normal FAB UI interface. The Jog page didn't show any error (in a previous installation a few days ago I had PHP errors) but I was unable to move the head (The error was saying that it didn't receive a 200 OK from the server).

When I checked for updates, it said it found one for Fablin because my currently installed version was `null`.

When I tried to reboot my unit, I got a kernel panic.

[image]

When I analyzed the partitions on the SD card using gparted, I could only see one partition, FAT32 labeled "BOOT", 512 MB large. The 4 MB before that partition and the 7.02 GB after it where shown as unallocated.

[image]

daniel-fabteam, 2017-03-04:
Take a look at the [sd card formatting guide](https://github.com/FABtotum/fabui-colibri/wiki/SD-card-formatting).

The 4MB before the partition is what's making the problem. The full issue has been solved here #20.

closed 2017-03-04 by daniel-fabteam

UnguidedFreedom, 2017-03-04:
I've already tried formatting my SD card from Linux and from MacOS X, exactly as per the SD card formatting guide and still have the issue (did it again just now).
I read issue #20 but couldn't find anything that would solve my problem.

daniel-fabteam, 2017-03-04:
Format your sdcard as it's described and check with gparted how the partition table looks. It must have only one FAT32 partition without any unallocated space before it. If it's not like that then continuing will for sure end up in a failure as the firstboot tools cannot resize the partitions properly and therefore partitions will be missing.

UnguidedFreedom, 2017-03-04:
I've checked using gparted, the partition generated on MacOS X (as per the sd card formatting guide) is actually preceded by a 4 MB unallocated space. I removed it, created a FAT32 partition on gparted that actually took all the space, put the files on it and the installation actually worked partition-wise.

[image]

However, I still wasn't able to jog (still that "didn't receive 200 OK from remote server" error), the update tool said Fablin version was `null`. But more importantly, after shutting down the unit, it can't boot anymore (not even to the gradient screen or to any text on the screen)...