FABtotum/fabui-colibri #63: Icons in input fields misaligned in during installation

seppestas, 2017-03-04:
The icons in the inputs fields of the settings menu shown during installation are misaligned with the input text, causing the icons to apear on top of the text:
[image]

## Environment
Chrome Version 56.0.2924.87 (64-bit)
OSX Version 10.12.3

krios-fabteam, 2017-03-06:
Issue fixed with the latest FABUI update

closed 2017-03-06 by krios-fabteam

seppestas, 2017-03-06:
👍 can it already be tested?

krios-fabteam, 2017-03-06:
Yes, just updating the new FABUI bundle or [downloading the latest sd card](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip)