FABtotum/fabui-colibri #64: Links to subsections on SD card formatting wiki page are incorrect

seppestas, 2017-03-04:
The links to the "Windows", "Mac" and "Linux" subsections link to the non-existing https://github.com/FABtotum/fabui-colibri/wiki/SDCard-formatting page instead of the https://github.com/FABtotum/fabui-colibri/wiki/SD-card-formatting page, causing the links to be redirected to https://github.com/FABtotum/fabui-colibri/wiki.

daniel-fabteam, 2017-03-04:
Thank you. Fixed.

closed 2017-03-04 by daniel-fabteam

seppestas, 2017-03-04:
Hey @daniel-fabteam, I also found a typo on https://github.com/FABtotum/fabui-colibri/wiki/Installation:

"Wait for the leds to became blue" in the last point should be "Wait for the leds to become blue" or "Wait for the leds to turn blue".

Not really worth an extra issue though imo :)

daniel-fabteam, 2017-03-04:
Also fixed. :)