FABtotum/fabui-colibri #65: Still OS corruption at boot time

florentmaye, 2017-03-06:
Half the time I boot my Fabtotum, I land on the recovery screen. I can then recove rcorrectly. I haven't identified any pattern leading to this situation. ( I had the impression that no ethernet cable plugged leads to that but in fact it seems it isn't the case)

daniel-fabteam, 2017-03-06:
Have you tried just to do **boot** instead of recovery?

You can land at the recovery page in one of three cases
1) Filesystem has been corrupted in such a way that automatic fixes cannot help
2) The head is removed or does not have a good contact
3) Both Y endstops have been pressed

florentmaye, 2017-03-06:
No never tried. I will next time.
I don't think I'm in #2 or #3 of these cases ...

daniel-fabteam, 2017-03-16:
Has this issue been resolved with the latest sdcard image and fw version?

closed 2017-03-27 by daniel-fabteam