FABtotum/fabui-colibri #67: buggy safety check

dr4Ke, 2017-03-07:
Last update (fabui 0.99.20170306.1805) introduced safety checks before printing (or they were very fast before) and they fails to detect my hybrid head, although its temperature is correctly reported (so I can tell that it is in fact connected).

dr4Ke, 2017-03-07:
I just updated to 0.99.20170307.1735. Have same symptoms. Sometimes, I also have "Bed inserted incorrectly", but it come back OK seconds later.
After a relatively long time, I'm redirected to the file selection screen. I suppose it's because the head is finally detected.

dr4Ke, 2017-03-07:
Here are the logs (from the system logs plugin)
[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/825131/archive.tar.gz)

dr4Ke, 2017-03-07:
Sometimes, it's good after a few seconds, other times, it's not after several minutes

krios-fabteam, 2017-03-09:
Issue fixed with the latest FABUI update

dr4Ke, 2017-03-09:
It's not better on my machine. In the `FabtotumServices.log` file, I have this:

    127.0.0.1 - - [09/Mar/2017 17:58:32] "POST /FABUI HTTP/1.0" 200 -
    DEBUG : put on command queue: M745,gcode
    DEBUG : << M745 [RQ: 0]
    DEBUG : __handle_line open [None]
    DEBUG :   >> [M745] [open]
    DEBUG : __handle_line ok [cmd: gcode, data: M745]
    DEBUG :   >> [M745] [ok]
    127.0.0.1 - - [09/Mar/2017 17:58:32] "POST /FABUI HTTP/1.0" 200 -
    DEBUG : put on command queue: M744,gcode
    DEBUG : << M744 [RQ: 0]
    DEBUG : __handle_line TRIGGERED [None]
    DEBUG :   >> [M744] [TRIGGERED]
    DEBUG : __handle_line ok [cmd: gcode, data: M744]
    DEBUG :   >> [M744] [ok]

So, for `M745`, it's not `TRIGGERED`, but `open`. Is this not normal?

I did the updates and a reboot before testing.

dr4Ke, 2017-03-09:
Wow. I just pushed on the head and it was not completely inserted. So it's all good. Sorry for the noise.

dr4Ke, 2017-03-09:
Talked too soon… It's correctly inserted, and still give `open` responses. Other commands are working as I can heat the nozzle, and it correctly reports the temperature.

dr4Ke, 2017-03-11:
I still have a "No head installed" with my hybrid head correctly inserted. I don't know what to do. Is this a bug in the software, or is this an error on my head, reporting `open` instead of `TRIGGERED`?

dr4Ke, 2017-03-11:
I think this is a firmware issue, as with versions before colibri, I have the `TRIGGERED` response.
So:

- good with Firmware v. 1.0.0097.2 and FabUI v. 0.986
- bad with Colibri version 0.99.20170310.1806 and firmware version 1.0.0097.2-rc2

The strange thing is:
Back from the old version to colibri, the `M745` command is still correctly responding `TRIGGERED`. After a firmware upgrade to 1.0.0097.2-rc2, it goes back to the bad `open` response. I tried to put back the 1.0.0097.2, manually downloaded from github, but it keeps responding `open`.

dr4Ke, 2017-03-11:
@daniel-fabteam @krios-fabteam Should I open a new issue?

dr4Ke, 2017-03-14:
@daniel-fabteam I still have this issue. I can't print anything with Colibri.

dr4Ke, 2017-03-15:
What can I do to help?

daniel-fabteam, 2017-03-16:
First switch to the latest fw version and make sure the pins are cleaned up.
Print heads check the serial (i2c) lines to see if the head is in place so it's possible to get temperature readings and still have the head not detected.

dr4Ke, 2017-03-17:
Latest software components installed. Firmware 1.0.0097.2-rc2. Pins cleaned. Same thing.

Tests:
- latest colibri + firmware 1.0.0097.2-rc2: `open`
- reboot to the latest pre-colibri release without flashing a new firmware: `TRIGGERED`
- reboot to colibri, still without changing firmware (still 1.0.0097.2-rc2): `open`

I didn't touch the head between each test. That's why I don't think that there is a hardware problem with my head and fabtotum.

dr4Ke, 2017-03-17:
Was there a change in the `M745` command that could explain the difference between the colibri version and the previous one?

dr4Ke, 2017-03-17:
I just realized that the head on my pre-colibri installation wasn't correctly configured. It was configured to be a Printing Head v2, not a Hybrid Head. I changed that, and I now have a more consistent behavior of `M745: open` in all cases. So I may take back my statement about not having a hardware problem.

Although, what's the difference in head detection for a hybrid head and a v2 head?
I can see in the FABlin code that it checks if all I2C reading are `63` it sets the head as not placed. How can I check myself these I2C reading?

daniel-fabteam, 2017-03-20:
There were no changes to `M745`. Could you make a fresh new sdcard with the rc2 firmware and see it you get the same output. 

I don't recall asking which unit version you have and which browser you are using.

dr4Ke, 2017-03-20:
I'm using the Fabtotum 1, from the 2013 indiegogo campaign, with RPi 1, and hybrid head v1.
Like I commented in my last message, there is no difference now between the 2 software versions, both have the same output for `M745` when configured for a hybrid v1 head.
I'm going to format my sdcard and try again now.

dr4Ke, 2017-03-25:
Tried with a formatted sdcard with latest image and updates. Same thing.
I tried something else. I exported the hybrid head parameters, changed its name and id (99 instead of 1) and imported this "new" head. Then I installed it and sent a `M745` command. The result is… `TRIGGERED`. That's comforting me in thinking that it's not a hardware problem.
Now, am I the only one to experience this issue?

So, I use a workaround: change hybrid head id.

simone-fabteam, 2017-03-27:
Hi @dr4Ke, thanks for your inputs and your proposed workaround. From the description it might be that your Hybrid Head does not respond correctly to I2C requests, so, yes, it would be a software related issue more than not. As one more check you can try entering the commands `M782` and `M784` when installed as Hybrid Head proper (ID: 1). For a correctly responding head they should output something along the lines of:

> M782
> Head Product ID: 70
> ok
> M784
> Head Serial ID: 000a9a1aa CRC: 84
> ok

For an I2C disabled head, they would give:

> M782
> Head Product ID: 3f
> ok
> M784
> Head Serial ID: 3f3f3f3f3f3f CRC: 3f
> ok

If they're all zeros instead, I2C has not been activated (such as when setting head ID to 99).

dr4Ke, 2017-03-27:
They are, indeed, responding `3f` every time, no matter the head ID (1 as the standard hybrid head, or 99 for my customed hybrid head definition). What that this mean?

```
M782
Head Product ID: 3f
ok
M784
Head Serial ID: 3f3f3f3f3f3f CRC: 3f
ok
```

simone-fabteam, 2017-03-29:
It's strange it gives '3f' in both cases. Basically the problem might be a wrongly flashed or corrupted firmware in the head. To confirm this, can you tell us whether you can correctly use the milling motor on your hybrid?

Moreover, for custom heads, please use ids from 100 onwards.

dr4Ke, 2017-03-29:
I don't know how to test the milling motor.

dr4Ke, 2017-03-29:
I tried a milling demo file, and, indeed, the milling motor is not working. What can I do?

dr4Ke, 2017-03-30:
Back to a newly formatted sdcard, with latest image, all updates, and finally, the latest firmware 1.0.0097.2-rc2.
Now, with a custom head (id 110) I have these answers:

```
M782
Head Product ID: 0
ok

M784
Head Serial ID: 000000 CRC: 0
ok
```
And with the classic hybrid head (id 1), i have:

```
M782
Head Product ID: 3f
ok

M784
Head Serial ID: 3f3f3f3f3f3f CRC: 3f
ok

M745
open
ok
```
But I still can't use the milling motor. Maybe I just don't know how to use it…

simone-fabteam, 2017-03-31:
It seems as if the head's firmware is corrupted. To simply test the milling function you can issue a `M3
S6000` from the jog page: if the heads emits a few buzzes and then start rotating the chuck it's ok, otherwise it's not. After that you can issue a `M5` to stop milling.

Anyway, I invite you to open a ticket on Fabtotum's support site (http://www.fabtotum.com/tickets) to receive more in-depth help for diagnosing this and evaluating possible solutions.

On our part, we'll take note of this to improve head diagnostic in fab-ui/colibri.

closed 2017-04-11 by daniel-fabteam