FABtotum/fabui-colibri #68: Dead on restart right after fresh install

christopherwit, 2017-03-08:
Just started the fabtotum successfully with fresh download of Colibri OS. Went trough setup and everything seems ok until I do a first reboot.
Something that might be relevant, I'm running my Fabtotum on a Pi3

Final error line posted during boot is at 11,5sec reading:
----{ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007e00

Let me know if you need additional data and how I can obtain this info for you.

krios-fabteam, 2017-03-09:
Hi, 
with which operating system did you formatted the sd card? Did you follow our [SD card formatting](https://github.com/FABtotum/fabui-colibri/wiki/SD-card-formatting) guide? If not could you please try to format the SD card by following it?

Take a look also at the close issue  #20

christopherwit, 2017-03-09:
Hi, fair chance it had something to do with the formatting as pointed out
in closed thread #20 <https://github.com/FABtotum/fabui-colibri/issues/20>
I'll give it a look and let you know about my progress.

christopherwit, 2017-03-09:
Hi Following the SD card formatting wizard did the trick. Currently happily restarting :)
Thanks and sorry for not searching around before posting an new issue. Maybe update the install instructions on [this page](http://www.fabtotum.com/fabui-colibri-world-first-os-3dprinting/) as well (as this is where I started).

krios-fabteam, 2017-03-09:
Don't worry @christopherwit , thanks for your contribution
Let's close the issue :)

closed 2017-03-09 by krios-fabteam