FABtotum/fabui-colibri #69: FABUI reporting "no internet" even though it is connected.

caliberza, 2017-03-09:
Since installing "colibri" my FAB reports "no internet connection" anytime I try access something that requires internet (i.e., check for updates).

The problem is the FAB is connected to the internet and it's working fine, I can ssh into it and access internet with no problem, so this appears to possibly be some kind of bug on the UI. I'm just not sure what I can do to further diagnose the issue?

For what it's worth, my FAB is configured with a custom IP/Subnet for my network, I also don't use WiFi.

daniel-fabteam, 2017-03-09:
Are you using the latest update?
We are releasing an update every day at 18:00. If you haven't updated colibri in a while do a new sdcard as there might have been changes that require a fresh system.

Can you share your network configuration? It might have something we didn't take into account.

caliberza, 2017-03-10:
I wasn't, but I just downloaded the latest image and the problem is still here.. 
I can shell in and "ping the internet" but the UI still reports no internet connectivity.

My network is configured as: Gateway (internet router) 192.168.0.1/24, my PC is 192.168.0.10/24, my Fabtotum has 192.168.0.9/24.

Also have a lot of other machines on the network (NAS box, family PC's etc) which is why I don't want to change anything.

daniel-fabteam, 2017-03-10:
When you say you can access the internet. How did you check internet works?

In case of static address you might want to check that DNS servers are configured properly too.
`ping 8.8.8.8` vs `ping www.google.com` will show if dns is working or not

caliberza, 2017-03-10:
Yea, I tested using ping. DNS is working fine as I can ping using domain names. (tried Google and a few others).

daniel-fabteam, 2017-03-15:
Make a new sdcard. In the latest updates we added **Disable** to wireless interface. Configure your eth network and disable the wireless interface. Internet should be detected.

If the issue is still there, re-open this issue.

closed 2017-03-15 by daniel-fabteam