FABtotum/fabui-colibri #71: Scan WiFi network button does not work

BramElderman, 2017-03-09:
Fab 1 RPi1 unit with buttonless wifistick,.. we got the wifi working by replacing the component however in the current build the scan option died and does not result in any detected network SSID
See log files enclosed
[archive(1).tar.gz](https://github.com/FABtotum/fabui-colibri/files/832322/archive.1.tar.gz)

BramElderman, 2017-03-09:
i will also try a rebuilt card from scratch

BramElderman, 2017-03-09:
The rebuilt card was (as once in the beginning) unable to create the USER mount so the partitioning seemed to  have failed... I was able to configure once but back to the kernel panic after reboot

bummer

daniel-fabteam, 2017-03-13:
Should be fixed in the latest update.

PS.: try to make a new sdcard with the one that was failing. Should work now.

closed 2017-03-15 by daniel-fabteam