FABtotum/fabui-colibri #75: tooltip for Z height incorrect

christopherwit, 2017-03-11:
Tooltip for z height increments has the following text, while default value is set at 0.5:
Do not set a value lower than 1(actual text might be slightly different as I translated from Dutch)

daniel-fabteam, 2017-03-13:
Could you specify in which view the tooltip is incorrect? Also I see you wrote "the following text" but don't see the text you are referring to.

Did you start working on Dutch translation?

christopherwit, 2017-03-14:
Sorry for the confusion. Hopefully below is more clear. 
Tooltip for z height increments on the jog page has the following text:
"Do not set a value lower than 1" (actual text might be slightly different as I translated from Dutch)
while default value is set at 0.5

Actually looking more into it, I'm surprised I do get a tooltip on that setting. I have English selected, and I didn't start translating:) Maybe in the future I will.
In addition I don't get tooltips on other input boxes, and as far as my testing did go it only appears in Firefox....

daniel-fabteam, 2017-03-16:
Is fixed in the latest update.
Note: It's required to make a new sdcard for the new update otherwise the feeder settings will not work properly.

closed 2017-03-16 by daniel-fabteam