FABtotum/fabui-colibri #76: 4th axis move prevented by cold extrusion warning

christopherwit, 2017-03-11:
4th axis move on jog panel is prevented by cold extrusion warnings even after executing a maintenance 4th axis engagement procedure. Only after clicking the turn off(cold extrusion warnings) button on the nozzle tab I can move the 4th axis freely.

daniel-fabteam, 2017-03-12:
Cold extrusion is a mechanism in Marlin firmware to prevent filament grinding if the nozzle is not hot enough to extrude. It's a 3D pritner after all. However as the feeder can be used as a 4th axis as well you can disable this safety mechanism while making sure the **feeder is disengaged** if you have a v1 unit or that you have **unloaded the filament** if your unit does not support feeder disengagement.

christopherwit, 2017-03-12:
Yes I'm running a Fabtotum v1 with disengagement mechanism.
So in both cases(unload filament or feeder disengage procedure) I need to manually disable cold extrusion warnings even after I executed the wizards? If so, please add for convenience those cold extrusion buttons on the 4th axis tab as well.

daniel-fabteam, 2017-03-14:
Is implemented in the latest update.

closed 2017-03-14 by daniel-fabteam