FABtotum/fabui-colibri #78: Nozzle height fine adjust wizard broken

christopherwit, 2017-03-11:
When I try to start nozzle height fine adjust procedure I end up with a splash screen telling me to please wait a bit longer and longer and longer ...
Nothing happens:)

daniel-fabteam, 2017-03-12:
Which version of the unit are you using?
Which OS did you use to format the sdcard?
And are you using the latest sdcard? There was a change in the calibration procedure in the latest releases so it's mandatory to create a new sdcard as an update will not work correctly.

christopherwit, 2017-03-12:
Download sdcard_latest [from this page](http://www.fabtotum.com/fabui-colibri-world-first-os-3dprinting/) on the 8th of march, plus all updates as provide by fabui yesterday evening.
I used Minitool partition wizard for Windows to format SD card.

Details from system info page are:
Serial number 43902-FAB-16372
Os Linux fabtotum 4.4.38-v7 #1 SMP Fri Feb 24 12:09:42 CET 2017 armv7l GNU/Linux
Fabui 0.99.20170310.1806
Firmware 1.0.0097.2-rc2 - Build date: Mar 10 2017 13:26:16 - Author: FABteam 
Hardware    1
Installed head Printing Head V2

Let me know if I already have the right sdcard_latest or I should reinstall a new download. So I can give an update later this week.

homahe, 2017-03-13:
I've noticed the same issue. 
To refine it a bit:
I'm using the same revision, same head and Fabtotum.
Calling  Jog  first, I get a 'Bed installed'  properly, and adjusting temperatures for head and bed is working fine.
If now I call Nozzle height calibration assisted, that's working too.
But calling Nozzle fine calibration which normally is just numerically adding or subtracting the correction value, I'm signalled to wait. Wait endlessly.. So I had to abort the operation.
If now I go to Jog, I get : No bed installed.
The bed is 'reinstalled' after a shutdown and restart.

BTW.:
After a print has finished, one cannot open the Maintenance menu ! I was able to open it only after having taken the detour over 'Dashboard'...

daniel-fabteam, 2017-03-13:
Should be fixed in the latest update. Best to make a new sdcard as there were some deeper changes when we introduced the new calibration procedure.

closed 2017-03-14 by daniel-fabteam