FABtotum/fabui-colibri #79: Pause problems

AndrewKillian, 2017-03-11:
When pausing the head dosen't move up away from the print so the nozzle melts the plastic already printed. Also the print will not always resume when told and takes a few presses of the button to resume.

homahe, 2017-03-13:
One example of how to use this:
If one wants a ball-bearing fixed in a bore, you can print a cylinder-bore with a stepping out from the inner diameter of the outer ring of the bearing to the outer diameter and stop (pause) at the height corresponding to the height of that ring, drop in the bearing, start printing again using the bearing as support...

Or if you have to print as step overhang, you can stop, put a support where the overhang starts and start printing again, without having trouble to first determine where to start printing and fix the support on the plate from the very beginning

jm2¢

daniel-fabteam, 2017-03-14:
The update that will be released today after 18h will move the head to safe position (top right corner) and the bed will go down 5cm. When resumed the head and bed will go back to previous position.

christopherwit, 2017-03-14:
I didn't test pause functionality yet, but what happens if the bed is already 3cm from the bottom when I press pause? I assume it is not going down 5cm right?

homahe, 2017-03-15:
No, 
just 3 cm and 1/10th up.

Best way to use 'pause' would be to be able to integrate it in the g-code. Otherwise you would have to sit waiting and watching, which wouldn't be feasible in a production code. And, yes, there's no real button to press, but to fiddle the  mouse-cursor into a tag position somewhere on the screen, which just now is not visible, because you scrolled up or down or are on another window. 
Besides it is hard to keep the cursor in a fixed position watching not the monitor but what is going on on the  building plate.  And there's no way to interact directly with the FABtotum, but to open the front door. 

In other words, to make it really useful it has to be integrated in the flow of the control-code (g-code) so it pauses automatically, signalling 'Caution, hot nozzle' as long as appropriate and switch of front door open - emergency (which should be reset to the previous state after an open/close cycle)
Ahh, and before I forget it, turn off the fan and heating not to waste energy, turn the heating on again after the front door is closed and the the fan on again when the nozzle is hot enough...

(I always wondered why the fan is turned on at 100%, when 'heating fast' during the nozzle adjustment, and fan actual state reading is 0%, target reading  0% but it is running full speed,
no button fan on/off, but you can pull the slider from zero to zero to switch it off ;-) )

daniel-fabteam, 2017-03-16:
It is already possible to pause the print using `M25` gcode.

closed 2017-03-21 by daniel-fabteam