FABtotum/fabui-colibri #81: How to trick no head detection for diy heads

christopherwit, 2017-03-12:
How can no-head-detection be tricked when a diy head without electronics is used and autorecovery shouldn't be initiated on boot up?
What kind of minimum electronics would be required?

daniel-fabteam, 2017-03-14:
The minimum for custom heads is to have a thermistor. If your custom head is not a print head then you can replace it with a regular 100K resistor.

christopherwit, 2017-03-14:
Sad to hear that several of my headmounts (pen holder, marker mount, swivelknife mount, etc) need a resister and I assume a decent type of contact(kind of pogo pin) from now on. But thanks for the clear answer.

closed 2017-03-14 by christopherwit

simone-fabteam, 2017-03-15:
Hi, as a workaround, if your heads hosts no electronics whatsoever, you can safely install them as a Milling Head V2. This, obviously, if having the FAB-UI believe it's doing milling works makes sense for your use cases.

Future releases of FABlin will provide the ability to configure minimum temperature checking and temperature sensing in more detail.

christopherwit, 2017-03-15:
Thanks for sharing this alternative. I look forward to test these
alterations you suggest.