FABtotum/fabui-colibri #82: How to install Linux packages

christopherwit, 2017-03-12:
As I'm completely new to colibri os and searching around Internet didn't answer my questions:
How can I install simple stuff like samba server, phpmyadmin and a gui on colibri os? Is a package manager available?

daniel-fabteam, 2017-03-12:
Colibri is a custom built OS with a single purpose of supporting FABUI and making it fault tolerant.

To extend the functionality of your printer you can use the plugin system which will get documentation after the beta testing phase.

What would you do with a samba server, phpmyadmin and a gui?

christopherwit, 2017-03-12:
Samba
For debugging and ease of use I would be able to browse the full system tree on another network attached device such as my windows pc downstairs. Previously it was a simple matter of installing samba. Is that even possible through a plugin system?

phpmyadmin
Well if the FABUI was flawless I wouldn't need access to edit project databases. In previous versions I used it multiple times to get things sorted(moving items between different projects without the need of uploading files again and again, after restoring a backup, edit update/ or cancelled print tasks that did get stuck in the process and to fix everything else that could go wrong while utilising a database)

GUI
I have a keyboard, mouse and screen attached to my pi3 running fabtotum, so I could control the printer from one desk without the need of an additional computer to access the webinterface in the raspbian based software. Please support as a single purpose your FABUI on a standalone unit as well by providing the methods to run a graphic webinterface on the Pi itself.

christopherwit, 2017-06-14:
As the response above wasn't very helping, I did research myself.

We have to cross compile binaries for armv71 architecture, as colibri is stripped down to a minimum and no package manager or other basic tools exist on this system to install even the most basic linux packages.