FABtotum/fabui-colibri #86: Nozzle hight calibration IMPOSSIBLE both up and down buttons only move the platform down

BramElderman, 2017-03-12:
My head kept on jamming in the platform when starting to print but when I do a Nozzle hight calibration setting a new hight is impossible since both buttons (up and down) only move the platform down... there is no way to calibrate...
For now the head jams into the platform at start,.. I quickly pauze and lower the platform since there is no way around...

NB when I asked if I want to store this new hight its does not store

krios-fabteam, 2017-03-16:
Which browser are you using?
Could you make a new fresh [SD card](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip) and check if the issue is fixed?

christopherwit, 2017-03-26:
I have the same issue in both IE11 and Firefox, but only when I set a move offset smaller then 0.5mm

AndrewKillian, 2017-03-26:
I'm having the same issue on Google Chrome

krios-fabteam, 2017-03-28:
It should be fixed in the last update
Once you confirm it's fixed close this issue.

krios-fabteam, 2017-03-29:
Does anybody can confirm this was fixed?

AndrewKillian, 2017-03-29:
I can confirm it's fixed

krios-fabteam, 2017-03-29:
Happy to hear it.
I close the issue

closed 2017-03-29 by krios-fabteam