FABtotum/fabui-colibri #88: Calibration disaster

homahe, 2017-03-13:
Hello,

using FABUI 0.9920170310.1806 / Firmware 1.0.0097.2
after having trouble with Nozzle Fine Calibration as stated 
the normal calibration did not work correctly, as the Z-distance when starting was less than the probe-length, so it could not switch out perpendicularly. The procedure was automatically repeated three times. 
Three times the probe did not reach 90°. Then the procedure stopped. (I'd guess it should either repeat and increase the Z-distance until the probe can switch out completely or travel down to the lower end-stop, switch out he probe and crawl up until the probe is reached.)
Nevertheless the calibration must have been changed during this operation, even when it failed, because when trying to print the head-distance became less zero.
Thereafter I wanted to do a complete new set-up, starting with the bed-calibration. 
This did not work: The bed travels down until the end-stop is hit, head having move to test-position 1travels upwards probe switched out, crawls up stops crawling before(!) the probe is reached. crawls down a little. Stops. Probe switched in, head to position 2, probe out, crawls up, probe not reached, so on thru pos 3 and 4. Head homing, probe switched in. Display reading:  Bed 100% correct 

Thereafter tried to reset the printer with no head installed. Started with bed-levelling, same behaviour.
(Q.: Are you counting the steps of the Z-motor after head crash and subtract the value form max Z-steps and use that as new max for Z-travel? If so, why is the max value not reset in the reset?)

daniel-fabteam, 2017-03-13:
Hi,

Please update your firmware to 1.0.0097.2-rc or go to **Maintenance -> Firmware** and select that version and flash it.

homahe, 2017-03-13:
Hi Daniel,

did as recommended, flashed to ..2-rc, restarted, checked version, bed 
calibration: The same problem: z-down, endstop, travels up,
probe never touched in all 4 positions, but WELL DONE...

daniel-fabteam, 2017-03-13:
When you type `M503` if jog console, what do you get for `probe_length` ?

homahe, 2017-03-13:
Result of M503:

M503
FABtotum TOTUMDUINO
 Batch Number: 3
FABlin
 Version: V 1.0.0097.2-rc2

Steps per unit:
  M92 X72.58 Y72.58 Z2133.33 E1552.60

Home offset (mm):
  M206 X0.00 Y0.00 Z0.00

Servo Endstop settings: R: 43 E: 145
Z Probe Length: -57.50

System_Log  follows, first have to install it

homahe, 2017-03-13:
Here's the system_log

[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/838378/archive.tar.gz)

daniel-fabteam, 2017-03-13:
Try this `M710 S38` in the jog console and re-run the nozzle calibration process before any other calibration.

homahe, 2017-03-13:
Hi Daniel,
I did enter the m710 S38  (nothing in the result window) then an 'Assisted nozzle calibration'
- down to endstop - up fast - 4 times: probe switched out, crawling up, probe not touched, crawling down, probe  switched in.
manual adjustment +/- offered, no movement triggered, neither up nor down.
Tried a second time:
Jog: m710 S38
Assisted  nozzle calibration, same procedure, but now the probe is touched each time, but the following manual adjustment is not working as before.
Tried a third time, now without m710 S38, (resetting) same result as before.

Nozzle offset calibration
====================================
New probe Length: 38.3625
New Z max offset: 241.5
New nozzle offset: 19.92

daniel-fabteam, 2017-03-13:
Could you use the **Emergency button**. Wait until it reloads the page. Check the probe_length is ~38. Rerun the failing calibrations and after all that download and attache the logs.

homahe, 2017-03-13:
You've got it, now calibration is working again.
Nozzle offset calibration
====================================
New probe Length: 38.5125
New Z max offset: 241.5
New nozzle offset: 4.98

Log:
[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/838622/archive.tar.gz)

daniel-fabteam, 2017-03-13:
Glad to hear it's working.

closed 2017-03-13 by daniel-fabteam