FABtotum/fabui-colibri #92: No print button in projectmanager on mobile device

christopherwit, 2017-03-14:
When viewing one of the objects in the projectmanager, the print button is rendered beyond my screen size and I'm unable to move the page around to get access to the print button. Even when I tap one of the printable items I do get the item page, but there is no action/print button as well.

My device is a rather small Sony xperia Z1 compact. 

Please find my screencapture attached below:
[image]

daniel-fabteam, 2017-03-15:
Try to rotate your device to see more of the table.

christopherwit, 2017-03-15:
Haha thanks for that bright suggestion :)
Just a small thing left: the pull-down menu renders (even with screen in landscape) of my screen:
[image]

krios-fabteam, 2017-03-16:
the fix will be available with the next update

closed 2017-03-20 by krios-fabteam