FABtotum/fabui-colibri #93: Bed leveling

lahines100, 2017-03-16:
When bed leveling command is selected the bed lowers to the bottom then raises to just below the print head. The head then moves to all four corners but never drops the ;probe and does not move the print bed. when complete says bed is perfectly level. Upon second calibrate, once again moves to all four corners, The bed moves up and down about 3mm but the probe never drops and nothing is measured. once again the bed is considered leveled. Latest updates are installed. System has been restarted twice.

daniel-fabteam, 2017-03-17:
We have changed the calibration procedure in the previous updates so it's needed to create a new sdcard. Check this issue #88 and see if the problem is resolved after a new sdcard.

PaoloP74, 2017-03-17:
I upgraded to the latest version via the update function.
I adjusted the height of the first nozzle, and then I switched to the flat setting.
Dish adjustment does not work. The plate reaches the position, the probe is put in position but the plate moves downwards.
After 1 cm lowering reverses direction and returns to the starting position. So for all the 4 point without that the touches the probe plate.
The result is 4 "wall done".
The second calibration (2 measures) is carried out in the same way but the head stops in the bottom right and then diagonally back to the starting point.

------------
Ho aggiornato all'ultima versione tramite la funzione update.
Ho regolato per primo l'altezza del nozzle e poi sono passato alla regolazione del piatto.
La regolazione del piatto non funziona. Il piatto raggiunge la posizione, la sonda si mette in posizione ma il piatto si muove verso il basso.
Dopo 1 cm di abbassamento inverte la direzione e torna in posizione di partenza. Così per tutti i 4 punto senza che il piatto tocchi la sonda.
Il risultato sono 4 "wall done".
La seconda calibrazione (2 misure) viene effettuata nello stesso modo ma la testa si ferma nell'angolo in fondo a destra e poi torna in diagonale al punto di partenza.

daniel-fabteam, 2017-03-17:
@PaoloP74 create a new sdcard instead of update.

lahines100, 2017-03-17:
Installed today's updates and issues 93 and 94 went away. current configuration is 
Serial number: 
Os: Linux fabtotum 4.4.38-v7 #1 SMP Fri Feb 24 12:09:42 CET 2017 armv7l GNU/Linux
Fabui: 0.99.20170317.1817
Firmware: 1.0.0097.2-rc2	- Build date: Mar 10 2017 13:26:16	- Author: FABteam
Hardware: 5
Installed head: either Hybrid Head or V2 print head

closed 2017-03-17 by lahines100