FABtotum/fabui-colibri #94: Nozzle Height Calibration

lahines100, 2017-03-16:
When doing the nozzle height calibration and CALIBRATE is selected, I receive the message "It seems thet you haven't done the nozzle height calibration for this head". and displays a calibrate button. I have the V2 print head installed.

daniel-fabteam, 2017-03-17:
Check issue #88.

lahines100, 2017-03-17:
Installed today's updates and issues 93 and 94 went away. current configuration is 
Serial number: 
Os: Linux fabtotum 4.4.38-v7 #1 SMP Fri Feb 24 12:09:42 CET 2017 armv7l GNU/Linux
Fabui: 0.99.20170317.1817
Firmware: 1.0.0097.2-rc2	- Build date: Mar 10 2017 13:26:16	- Author: FABteam
Hardware: 5
Installed head: either Hybrid Head or V2 print head

closed 2017-03-17 by lahines100