FABtotum/fabui-colibri #100: reboot is required after first modification of eth ipnr

christopherwit, 2017-03-17:
Internet doesn't get activated directly after saving the new static IP number for ethernet. The new IP number is functional, and the page is refreshed to the new IP number. Only internet is still down. After reboot everything is fine.

daniel-fabteam, 2017-03-20:
Did you check the internet by logging in over ssh or was it not detected in the UI?

christopherwit, 2017-03-20:
After changing IPnr and the redirect I can also access the Fabtotum over ssh successfully.
The only thing where I can tell internet isn't working is because web feeds aren't loaded and internet connection icon is missing.
As said after reboot everything is in good order.

christopherwit, 2017-03-26:
Still the case in latest image.

daniel-fabteam, 2017-03-31:
Will be fixed in the next update. (after 18.00 GMT+1)
Once you confirm it is fixed, please close this issue.

christopherwit, 2017-03-31:
except for the internet icon which doesn't come up after altering the settings, the page gets refreshed and all social feeds load correctly.

closed 2017-03-31 by christopherwit