FABtotum/fabui-colibri #103: abort button during PID calibration doesn't work

christopherwit, 2017-03-17:
In addition at least once my fabtotum completely freezed when I selected to reset controller during PIDtuning. Even after reboot it keeps blinking yellow, emergency and controller reset didn't had any impact. During the following reboot the fabtotum didn't boot anymore as it was stuck at closing partitions(I think).  After manual reboot I discovered a firmware issue, as the installed version was listed as null. (but no update highlighted by an indication in the main menu).
Updating to new firmware is stuck at connecting download server.
To be continued as I don't want to continue this story today. Maybe on Sunday I will give it another shot by going through a recovery and see if this fixes the issue with my controller. Any recommendations for during the recovery procedure?

christopherwit, 2017-03-19:
I did go through a full recovery as I wasn't sure which check was required to get the totumdino reflashed.
The printer did got fixed, but it wasn't me if I wanted to be sure the firmware issue was caused by pid tune abort, so I tried again.
Aborting pid tune does give issues with taskpane update, and fiddling around with abort and page refrash doesn't solve the issue. When I tried to do a controller reset again everything got stuck. The splashscreen shows pid autotune started, nozzle temp set to 200, set to 50 and the wait circle keeps spinning.
Recovery boot up was again the only way getting the printer back up.

christopherwit, 2017-03-26:
Not so may issues as before, but in latest update PID tune abort button still doesn't stop PID tune procedure and reset temperature to 0degC.

daniel-fabteam, 2017-04-10:
The fix will be available in today's update after 18:00 (GMT+1).

If it is resolved, please close this issue.

christopherwit, 2017-04-10:
Fix confirmed.

closed 2017-04-10 by christopherwit