FABtotum/fabui-colibri #97: allmost any page on http:\\ipnr\recovery don't work

christopherwit, 2017-03-17:
Several of the recovery pages accessed by ipnr\recovery are not working. I do realise these recovery pages are maybe outdate, but I just wanted you to not forget about them (by removing or fixing).
Most of them have a lot of 404 errors. For all remarkable stuff I have made a shortlist:
-recovery\wlan.php renders a page showing completely different settings than the fabui network page
-recovery\eth.php same as above
-recovery/info.php?mode=fw blank page
-recovery/info.php?mode=raspi funny enough it pops up a pi cam video stream on my printer attached screen
-recovery/log.php?type=kernel page without log content
-recovery/flash.php no content
-recovery/setup.php internal server error 500
-phpmyadmin 404 page not found
-recovery/macrosim.php no sensible content
-recovery/test.php buttons don't work properly, page does miss some content
-recovery/install the only working page which brings me to welcome wizard

krios-fabteam, 2017-03-20:
hi @christopherwit 
the "/recovery/*" pages are obsolete now and will not be imported to fabui-colibri. You will find the  main functionalities of these pages in "maintenace" and "settings" sections, in additional also more plugins will be available

christopherwit, 2017-03-20:
Please remove these obsolete pages in that case.

closed 2017-03-21 by krios-fabteam