FABtotum/fabui-colibri #104: Print issues with colibri

mgcrea, 2017-03-18:
Just took out the fabtotum out of its box to try this. And I've been really pleased with the results, the UI is way more snappier and nice than the old one. Calibration worked directly.

However I'm encountering issues with printing that I never had with the legacy OS. Somehow the head gets stuck and the motor makes unusual loud high pitch noises that it did not do before, for instance for a marvin keychain that used to work fine on the previous UI. First print was the spool lever that went somehow OK (print detached from bed du to too low bed temp @ 40C).

Here is a quick vid, https://youtu.be/gu6juxC0yy0, you'll hear the loud noises.

Thanks for the awesome work.

mgcrea, 2017-03-18:
I've tested some old files, and it looks like I still get the old behavior. So the issue lies probably in the new samples. That are somehow incompatible with my machine?

daniel-fabteam, 2017-03-20:
@mgcrea Hi, which unit version do you have? You can check this in **Maintenance -> System info**

mgcrea, 2017-03-21:
```
Serial number
D3AC5-FAB-99E40
Os
Linux fabtotum 4.4.38-v6 #1 Fri Feb 24 12:02:46 CET 2017 armv6l GNU/Linux
Fabui
0.99.20170321.1832
Firmware
1.0.0097.2	- Build date: Mar 1 2017 12:36:03	- Author: FABteam
Hardware
1
Installed head
Printing Head V2
```

daniel-fabteam, 2017-03-22:
Could you update the fw to the latest `1.0.0097.2-c2` version?

After that go to jog control and type 

    M502
    M503

And do an emergency stop. The triangle button in the top right corner.

Then try the sample that was behaving wrong.

If this does not help. Go to project manager, click on the sample to get tot the editor and remove any `M20x` lines from the beginning and try to print it again.

mgcrea, 2017-03-25:
```
M502
echo:Hardcoded Default Settings Loaded
ok
M503
FABtotum TOTUMDUINO
 Batch Number: 1
FABlin
 Version: V 1.0.0097.2-rc2
 Baudrate: 250000
Steps per unit:
  M92 X72.58 Y72.58 Z2133.33 E3048.16
Maximum feedrates (mm/s):
  M203 X250.00 Y250.00 Z15.00 E12.00
Maximum Acceleration (mm/s2):
  M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
  M204 S4000.00 T100.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
  M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm):
  M206 X0.00 Y0.00 Z0.00
PID settings:
   M301 P15.00 I5.00 D30.00
Servo Endstop settings: R: 25 E: 127
Z Probe Length: -31.65
Installed head ID:
 M793 S2
ok
```

I've done an assisted nozzle height calibration after that (not sure if that was necessary).

And... That fixed it! Thanks!

closed 2017-03-25 by mgcrea

mgcrea, 2017-03-25:
As a footnote, the printing of the marvin keychain failed due to a bed temp too low (35°C) as the legs did detach from the bed after a few layers. Maybe thoses temps should be bumped a bit (I usually get no issues at 60°C.

mgcrea, 2017-03-31:
@daniel-fabteam just to let you know that I had the same issue after flashing a new sd card. So you might want to factory reset the head after the head is selected in the GUI.