FABtotum/fabui-colibri #110: extruder temp critical

mjb984, 2017-03-22:
[extruder temp error.docx](https://github.com/FABtotum/fabui-colibri/files/860299/extruder.temp.error.docx)

please help.

after latest update i get the following error when ever i attempt to use the 3d extruder v2. this occurs ay time i try to heat it up. can't load filament, jog, or print.

please advise.

daniel-fabteam, 2017-03-22:
Could you provide your unit version and browser/os you are using?
The unit version can be found under **Maintenance -> System Info**

Also type `M801` in the jog control and paste the output here. This command will show what's the configured max temperature.

daniel-fabteam, 2017-03-22:
Should be fixed in the latest update.

daniel-fabteam, 2017-03-29:
@mjb984 Could you confirm that the issue is fixed and if so close it.

mjb984, 2017-03-30:
sorry, yes. i was able to resolve the issue using the PID option in nozzle settings.

closed 2017-03-30 by mjb984