FABtotum/fabui-colibri #114: Feeder calibration altered value stored without notice

homahe, 2017-03-23:
With FABUI 0.99.20170321.1832:

After installing the previous update the feeder value had changed, so the prints came out off shape. Therefore the feeder had to be re-calibrated. Instead of 100 mm 185 mm were delivered. 
The result for the new value shown was 1647,6450540541. 
Remarkable: First the "precision" of the decimal value, second it is shown with decimal comma instead of decimal-point as usual.
If now you don't save the result, but start the calibration again, the new value (1647,64...) is nevertheless stored as reference without notice, as now a new value is shown, 890,62388130222, whereas neither the 100mm to feed but 185 mm fed had changed, nor the previous result was explicitly stored (saved).

Overriding the value by entering the corrected value and trying to save it explicitly using the button provided was canceled after waiting for more than 10 minutes for the procedure to finish.

Using the 'Profiles' menu, entering the correct value did work.

krios-fabteam, 2017-03-23:
Could you please update FABUI to the latest version (current is 0.99.20170323.1742) and confirm the issue is fixed? If not try to flash SD card again with the [latest archive](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip)

homahe, 2017-03-23:
using FABUI release 0.99.2017.03233.1742 doing the step calibration, I get the following:
old value:1647,65
extract 100 mm filament:  true length   101  calculated value 1599.3496408465
I did not change the value, 
extract 100 mm filament: true length   102
reading now in actual value 1599.3496408465
calculated value shows: 1599.3496408465
hit calculate:
calculated value changes to 1567.9898439672
actual value  changes to 1567.9898439672 

Extracting 100 mm a third time: true length 103 mm

Calibration completed New value: 1522.3202368614
Entering value 1522,20 using the manual change value (here decimal point is comma..)
Start extrusion 100mm with actual steps 1522.2
getting 95 mm
calculate -> 1602.3157894737 steps -> actual value 1602.3157894737
getting 95 mm again(!)
calculate -> 1686.648199446 steps -> actual value 1686.648199446
getting 95 mm again(!)
change value manually to 1602.3157894737  storing  perpetual..
go to Profiles  -> head V2 -> edit:  Steps per Unit  1602,36 (comma here too)
test calibration: result  100mm = 100 mm
DONE.

krios-fabteam, 2017-03-28:
Fixed with the latest update
Once you confirm it's fixed close this issue.

krios-fabteam, 2017-04-03:
If it still happens, please re-open this issue

closed 2017-04-03 by krios-fabteam