FABtotum/fabui-colibri #116: Bed calibration after emergency stop causes havoc

homahe, 2017-03-23:
FABUI 0.99.2017.0321.1832

Situation:

Starting a print I noticed the bed wasn't properly leveled and so I opened the front door to stop the print, as aborting takes longer and one has to roll, point and click.. So now we've an emergency stop. 

Regaining access and starting  a bed calibration, the head starts to move just as if  starting at home-position, resulting in an override of  x-Max.

If now one switches of power and restarts the calibration without manually homing the head via jog the bed isn't raised up to touch the probe, if it was to deep when switching off power, but the four positions are traveled and the result is a "100%  leveled bed" . 

Solution: Do a full homing (via jog) before starting the bed-leveling procedure.

krios-fabteam, 2017-03-27:
Full homing is  one of the macro executed before the bed calibration starts, did you updated FABUI to the latest version and check if the issue was fixed (it should be)

krios-fabteam, 2017-03-30:
hi @homahe, 
are you still experiencing this issue?

krios-fabteam, 2017-04-03:
This should be fixed with the latest updates
If it still happens, please re-open this issue

closed 2017-04-03 by krios-fabteam