FABtotum/fabui-colibri #119: Feeder Problems

AndrewKillian, 2017-03-24:
all other functions in jog work however when I try to jog out some filament nothing happens. Also I tired to load a spool of filament and the printer just made terrible high pitch whining noises so I emergency stopped. Before this the filament wouldnt grab so I was able to remove the spool and found it had been ground down at a spot as well. I'm not sure if there's any coorelation but it has also been making a simiar sound when it probes before prints.

daniel-fabteam, 2017-03-27:
Try executing following commands in jog control.

    M502
    M500

And then do an emergency stop. This should reset every fw setting to default and should fix the high pitch issue.

Attach the logs that you can get from the **System Logs** plugin.

krios-fabteam, 2017-04-03:
@AndrewKillian are you still experiencing this issue?

AndrewKillian, 2017-04-03:
not since the recent updates

closed 2017-04-03 by AndrewKillian