FABtotum/fabui-colibri #122: Print height very far off

AndrewKillian, 2017-03-28:
After using the assisted calibration, the height is calibrated fairly well, requiring adjustments of usually at most .2mm, however, on the following print the height hasn't even been close needing adjustments of upwards of 4.5mm. We had this issue of print heights needing adjustment almost every print but they have only been this extreme after the last two updates.

krios-fabteam, 2017-03-29:
Are you doing the assisted calibration with the probe enabled? If yes try to disable it in "Settings>Hardware>Homing" and try again the calibration.
And wich firmware version do you have installed? Please paste here M503 output (you can  find it in "Maintenance>System Info"

mgcrea, 2017-03-30:
Not sure if I'm encountering the same issue but I've setup a new sd-card today (last one died), and after a nozzle height calibration my first print was a bit far off, so I overriden the Z-height (`Override Z height: -0.20`) and it printed OK.

After that I went to manually fine calibrate the nozzle height to reduce by `0.2` but it somehow failed to register for the next print (still had to override -0.20 to properly print). Isn't the nozzle supposed to fix it? Thought there was a direct correlation between the two (nozzle height & z-height).

krios-fabteam, 2017-03-30:
Are you guys doing the nozzle height calibration with probe enabled?

mgcrea, 2017-03-30:
I'm not using the probe. Only assisted with a sheet of paper (and sometimes manual to adjust an offset).

AndrewKillian, 2017-03-30:
No I'll try with the probe when I get home

krios-fabteam, 2017-03-30:
We've noticed an issue with probe management so before your test please update the FABUI (as usual we'll release a new version after 18.00 GMT+1 )

krios-fabteam, 2017-04-03:
@AndrewKillian can we consider this issue as resolved?

AndrewKillian, 2017-04-03:
For now I'll bring it up again if there are further issues

closed 2017-04-03 by AndrewKillian